MrBuddyCasino/ESP32_MP3_Decoder #1: Built-In DAC does not work

MrBuddyCasino, 2017-01-22:
Running without the I2S codec can be activated by uncommenting "#define USE_DAC", but this does not currently work.

If anyone has experience using I2S_MODE_DAC_BUILT_IN, please have a look!

negativekelvin, 2017-01-22:
No experience but i2s_set_pin
https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/blob/master/main/app_main.c#L304
Looks like should be called with null value
https://github.com/espressif/esp-idf/blob/master/components/driver/i2s.c#L390

MrBuddyCasino, 2017-01-23:
Good catch! Will try that out.

MrBuddyCasino, 2017-01-23:
Makes a sound now, but nothing you'd want to listen to.

negativekelvin, 2017-01-24:
Hmm have you tried to use only I2S_MODE_DAC_BUILT_IN flag

MrBuddyCasino, 2017-01-24:
Yes, look at he init_i2s_dac() function in app_main.c. Its doing exactly what the sample code in i2s.h says we should do:

` * ----------------EXAMPLE USING I2S WITH DAC ---------------------
 * @code{c}
 *
 * #include "freertos/queue.h"
 * #define I2S_INTR_NUM 17                                      //choose one interrupt number from soc.h
 * int i2s_num = 0;                                             //i2s port number
 * i2s_config_t i2s_config = {
 *    .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN,
 *    .sample_rate = 44100,
 *    .bits_per_sample = 8,                                    // Only 8-bit DAC support
 *    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,            //
 *    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
 *    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
 *    .dma_buf_count = 8,
 *    .dma_buf_len = 64
 * };
 *
 *
 * i2s_driver_install(i2s_num, &i2s_config, 0, NULL);   //install and start i2s driver
 *
 * i2s_set_pin(i2s_num, NULL); //for internal DAC
 *
 * i2s_set_sample_rates(i2s_num, 22050); //set sample rates
 *
 * i2s_driver_uninstall(i2s_num); //stop & destroy i2s driver
 *@endcode`

Spritetm, 2017-01-29:
Fyi, it seems the I2S driver has a bug in DAC mode: the sample rate it sets is actually 8 times the sample rate you tell it to use. I have a bug fix in the internal merge queue, it will probably be merged after the holidays.

glennhuber, 2017-02-13:
When will this bug fix be available?

MrBuddyCasino, 2017-02-14:
The SDK fix hasn't landed yet, but in theory you can workaround the issue by modifying set_dac_sample_rate(int rate) and init_i2s_dac() to divide the sample rate by 8. Haven't tested that, though.

MrBuddyCasino, 2017-04-13:
Supersat's changes indeed fix this issue. I2S codec no longer strictly required. :-)

closed 2017-04-13 by MrBuddyCasino