MrBuddyCasino/ESP32_MP3_Decoder #3: port numbers

jobidon, 2017-03-25:
Thanks for making this available!
When testing, I had to make the following changes:
- In playerconfig.h, each of the port numbers should be specified as strings (adding the double quotes before and after)
- Modify the pin numbers for the i2s module in app_main.c line 269~271, making sure you use an output pin for data out (gpio<33)

MrBuddyCasino, 2017-03-29:
> Modify the pin numbers for the i2s module in app_main.c line 269~271, making sure you use an output pin for data out (gpio<33)

Not sure I got what you mean? I'm using the default I2S pins afaik.

jobidon, 2017-03-29:
just wanted to point out that gpio pins 33 and above are inputs only. To provide a signal to the module, one must use pins 32 or below.

MrBuddyCasino, 2017-03-29:
I just pushed a big update, stations are now entered as simple URLs in playerconfig.h.
I2S pins are: bck = 26, ws = 25, data = 22

closed 2017-03-30 by MrBuddyCasino