MrBuddyCasino/ESP32_MP3_Decoder #6: Built-In ADC can't work with latest ESP-IDF

tidyjiang8, 2017-05-08:
Hi Michael,

I reviewed the history, and think it should work for Built-In DAC case, but when I tried to run the latest code with latest ESP-IDF, I failed. 

It seems that it's the version of ESP-IDF that does not match with this code. To be more specific, it's the version of I2S driver of ESP-IDF that does not match. In the latest I2S driver, it does not support the 8 bit width per sample. 

I don't know why and how, could you tell me which version of ESP-IDF you are using? Or could you have a check at the latest ESP-IDF?

The version of ESP-IDF I'm using is `07b61d5 (2017-04-28)`

Thanks & Best Regards,
Tidy Jiang.

MrBuddyCasino, 2017-05-08:
I'm always on the master branch, latest commit a41ac2d21d90fc69f8426eb6036d2d1e1274ec28. Did you active the DAC workaround option? Its currently required to make it work.

tidyjiang8, 2017-05-08:
Attach the log.
```
Hardware initialized. Waiting for network.
I (15654) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (16644) wifi: state: init -> auth (b0)
I (16644) wifi: state: auth -> assoc (0)
I (16654) wifi: state: assoc -> run (10)
I (16674) wifi: connected with test31, channel 6
I (17674) event: ip: 192.168.1.120, mask: 255.255.255.0, gw: 192.168.1.1
I (17674) mDNS: starting mDNS
I (17674) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (17684) renderer: init DAC_BUILTIN on port 0
E (17684) I2S: Invalid bits per sample
I (17694) main: RAM left 66992
I (18254) http_client: DNS lookup succeeded. IP=173.239.76.148
I (18254) http_client: ... allocated socket
I (18464) http_client: ... connected
I (18474) http_client: requesting GET /illstreet-128-mp3 HTTP/1.0
Host: ice1.somafm.com

I (18474) http_client: ... socket send success
Buffer fill 25%, 16012 bytes
Buffer fill 49%, 31678 bytes
Buffer fill 74%, 47568 bytes
I (21994) audio_player: RAM left 68992
I (21994) audio_player: created decoder task: mp3_decoder_task
MAD: Decoder start.
dec err 0x0235 (bad main_data_begin pointer)
dec err 0x0235 (bad main_data_begin pointer)
I (22014) renderer: setting sample rate to 2756

E (22014) I2S: /home/work/esp/esp-idf/components/driver/./i2s.c:609 (i2s_set_sample_rates):bits_per_sample not set
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40118d2f  PS      : 0x00060130  A0      : 0x801170a8  A1      : 0x3fff1800  
0x40118d2f: i2s_push_sample at /home/work/esp/esp-idf/components/driver/./i2s.c:910

A2      : 0x00000000  A3      : 0x3fff1830  A4      : 0x00000005  A5      : 0x00000000  
A6      : 0x0001a03c  A7      : 0x00000066  A8      : 0x00000080  A9      : 0x3fff1850  
A10     : 0x0000007e  A11     : 0x3ffde2f0  A12     : 0x3fff1890  A13     : 0x0000000c  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40118d2f:0x3fff1800 0x401170a8:0x3fff1830 0x4013a56c:0x3fff1860 0x4013a614:0x3fff1920 0x4011e3b1:0x3fff1940
0x40118d2f: i2s_push_sample at /home/work/esp/esp-idf/components/driver/./i2s.c:910

0x401170a8: render_sample_block at /home/work/esp/ESP32_MP3_Decoder/components/audio_renderer/./audio_renderer.c:122

0x4013a56c: synth_full at /home/work/esp/ESP32_MP3_Decoder/components/mad/./synth_stereo.c:743 (discriminator 2)

0x4013a614: mad_synth_frame at /home/work/esp/ESP32_MP3_Decoder/components/mad/./synth_stereo.c:938

0x4011e3b1: mp3_decoder_task at /home/work/esp/ESP32_MP3_Decoder/components/mp3_decoder/./mp3_decoder.c:140

Rebooting...
```

tidyjiang8, 2017-05-08:
Yes, I have activated workaround.

MrBuddyCasino, 2017-05-08:
I just pushed a fix. DAC mode works again, but the audio is distorted. I'll have to look into this a bit more.

tidyjiang8, 2017-05-08:
Thank you for your enthusiasm and looking forward to see your new  change.

MrBuddyCasino, 2017-05-08:
I think I fixed it now, I'll clean up the obsolete code later. Thanks for the bug report.

closed 2017-05-08 by MrBuddyCasino