MrBuddyCasino/ESP32_MP3_Decoder #7: always single channel(left)

aifer2007, 2017-05-13:
I use the I2S mode， but only left channel works， right channel outputs nothing.

My  DAC is es9018K2M chip.

MrBuddyCasino, 2017-05-13:
I usually have a mono amp connected, so it is entirely possible I missed that - will check.

aifer2007, 2017-05-14:
Thanks.

And , In the DAC_BUILT_IN mode, a mp3 file can be playbacked, but in I2S mode, bit_depth = I2S_BITS_PER_SAMPLE_16BIT, sample_rate_modifier = 1, the same mp3 file will be playbacked with large noise. I must use I2S mode, bit_depth = I2S_BITS_PER_SAMPLE_32BIT, sample_rate_modifier = 2 to palyback the file.

MrBuddyCasino, 2017-05-14:
Can you tell me the 
- sample rate 
- number of channels 
of the MP3 file?

aifer2007, 2017-05-14:
Yes.

sample rate：44100
channel num： 2

MrBuddyCasino, 2017-05-15:
I've pushed some fixes - can you try again?

I've used the ES9018K2M in the past and found it to be a pit picky, you might want to have a look at your signals if you have an oscilloscope. I2S 16 Bit mode should work as intended.

aifer2007, 2017-05-15:
Thanks. 

I upgrade to the latest codes, Try it on ESP32 and ES9018K2M DAC：

I2S mode can playback, but noise is as large as before.

I2S_MERUS mode works well. （It doesn't work before）

DAC_BUILT_IN mode doesn't work. MAD starts, but no sound output from the speaker, no noise also.

MrBuddyCasino, 2017-05-15:
Just tested again:

- I2S, 16 Bit: works
- I2S, 32 Bit: works
- PDM: works
- DAC: doesn't work

I have no idea whats going on with DAC mode - I don't remember changing anything. I suspect I2S/16Bit might be an issue with your chip, but I'll check again as soon as I have a stereo amp.

In the meantime, use PDM if you want to go DAC-less, it sounds better anyway.

MrBuddyCasino, 2017-05-15:
Ok found the bug - DAC works again.

aifer2007, 2017-05-15:
In old codes, DAC mode is 8bit， it can works. Now DAC mode is 16bit, doesn’t work.

And, I’m not family with pdm, can you give me some help? for example, connections?

aifer2007, 2017-05-15:
Great!

Can you give me some help on PDM？

MrBuddyCasino, 2017-05-15:
> In old codes, DAC mode is 8bit， it can works. Now DAC mode is 16bit, doesn’t work.

I'm always on the latest master branch version of the ESP-IDF, they changed how the DAC works, so you need to update your SDK.

> Can you give me some help on PDM
You can use it just like the DAC - connect a speaker to ground and pin 25 or 26. If you add a low-pass filter, quality could be really good. Theres a schematic in the project's /doc folder that gives you some hints what to do.

aifer2007, 2017-05-15:
Some mp3 file will make the decoder crash.

The file is too large to send to you.

```
I (6822) http_client: requesting GET /cg.mp3 HTTP/1.0
Host: 192.168.1.9

I (6822) http_client: ... socket send success
I (6922) audio_player: Buffer fill 50%, 16066 bytes
I (6982) audio_player: RAM left 175916
I (6982) audio_player: created decoder task: mp3_decoder_task
MAD: Decoder start.
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0102 (reserved header layer value)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
I (7002) audio_player: Buffer fill 70%, 22647 bytes
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0104 (reserved sample frequency value)
dec err 0x0104 (reserved sample frequency value)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
E (7052) decoder: Buffer underflow, need 2881 bytes.
E (7062) decoder: Buffer underflow, need 2881 bytes.
E (7062) decoder: Buffer underflow, need 2881 bytes.
E (7072) decoder: Buffer underflow, need 2881 bytes.
E (7072) decoder: Buffer underflow, need 2881 bytes.
E (7082) decoder: Buffer underflow, need 2881 bytes.
E (7082) decoder: Buffer underflow, need 2881 bytes.
E (7092) decoder: Buffer underflow, need 2881 bytes.
E (7102) decoder: Buffer underflow, need 2881 bytes.
E (7102) decoder: Buffer underflow, need 2881 bytes.
E (7112) decoder: Buffer underflow, need 2881 bytes.
E (7112) decoder: Buffer underflow, need 2881 bytes.
E (7122) decoder: Buffer underflow, need 2881 bytes.
E (7122) decoder: Buffer underflow, need 2881 bytes.
E (7132) decoder: Buffer underflow, need 2881 bytes.
E (7142) decoder: Buffer underflow, need 2881 bytes.
E (7142) decoder: Buffer underflow, need 2881 bytes.
E (7152) decoder: Buffer underflow, need 2881 bytes.
E (7152) decoder: Buffer underflow, need 2881 bytes.
E (7162) decoder: Buffer underflow, need 2881 bytes.
E (7162) decoder: Buffer underflow, need 2881 bytes.
E (7172) decoder: Buffer underflow, need 2881 bytes.
E (7182) decoder: Buffer underflow, need 2881 bytes.
E (7182) decoder: Buffer underflow, need 2881 bytes.
E (7192) decoder: Buffer underflow, need 2881 bytes.
E (7192) decoder: Buffer underflow, need 2881 bytes.
E (7202) decoder: Buffer underflow, need 2881 bytes.
E (7202) decoder: Buffer underflow, need 2881 bytes.
E (7212) decoder: Buffer underflow, need 2881 bytes.
E (7212) decoder: Buffer underflow, need 2881 bytes.
E (7222) decoder: Buffer underflow, need 2881 bytes.
E (7232) decoder: Buffer underflow, need 2881 bytes.
E (7232) decoder: Buffer underflow, need 2881 bytes.
E (7242) decoder: Buffer underflow, need 2881 bytes.
E (7242) decoder: Buffer underflow, need 2881 bytes.
E (7252) decoder: Buffer underflow, need 2881 bytes.
E (7252) decoder: Buffer underflow, need 2881 bytes.
E (7262) decoder: Buffer underflow, need 2881 bytes.
E (7272) decoder: Buffer underflow, need 2881 bytes.
E (7272) decoder: Buffer underflow, need 2881 bytes.
E (7282) decoder: Buffer underflow, need 2881 bytes.
E (7282) decoder: Buffer underflow, need 2881 bytes.
E (7292) decoder: Buffer underflow, need 2881 bytes.
E (7292) decoder: Buffer underflow, need 2881 bytes.
E (7302) decoder: Buffer underflow, need 2881 bytes.
E (7302) decoder: Buffer underflow, need 2881 bytes.
E (7312) decoder: Buffer underflow, need 2881 bytes.
E (7322) decoder: Buffer underflow, need 2881 bytes.
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0104 (reserved sample frequency value)
dec err 0x0101 (lost synchronization)
dec err 0x0201 (CRC check failed)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0102 (reserved header layer value)
dec err 0x0101 (lost synchronization)
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x8011e147  A1      : 0x3ffd7490  
A2      : 0x3ffd7554  A3      : 0x3ffc35d8  A4      : 0x3ffda9a4  A5      : 0x0000007f  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8015cea4  A9      : 0x3f43d9a8  
A10     : 0x3ffc35d8  A11     : 0x3ffd7554  A12     : 0x00000050  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x0000045a  SAR     : 0x00000018  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffd7490 0x4011e147:0x3ffd74c0

Rebooting...
ets Jun  8 2016 00:22:57

```

aifer2007, 2017-05-15:
cg.mp3 info:
bit rate: 192 kbps
sample rate: 44.100 kHz
id3 : v2.3
code format : Lavf56.4.101

aifer2007, 2017-05-15:
May I send the mp3 to you?

MrBuddyCasino, 2017-05-15:
Also please raise separete issues for new bugs, that way its easier for me to keep track of everything.

MrBuddyCasino, 2017-05-15:
> May I send the mp3 to you?

Sure, just upload it to Google Drive or something. You can send the link to buschfunk@yahoo.com if you don't want to write it here.

closed 2017-05-15 by MrBuddyCasino