MrBuddyCasino/ESP32_MP3_Decoder #8: Some mp3 file will make the decoder crash.

aifer2007, 2017-05-15:
Some mp3 file will make the decoder crash.

```
I (6822) http_client: requesting GET /cg.mp3 HTTP/1.0
Host: 192.168.1.9

I (6822) http_client: ... socket send success
I (6922) audio_player: Buffer fill 50%, 16066 bytes
I (6982) audio_player: RAM left 175916
I (6982) audio_player: created decoder task: mp3_decoder_task
MAD: Decoder start.
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0102 (reserved header layer value)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
I (7002) audio_player: Buffer fill 70%, 22647 bytes
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0104 (reserved sample frequency value)
dec err 0x0104 (reserved sample frequency value)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
E (7052) decoder: Buffer underflow, need 2881 bytes.
E (7062) decoder: Buffer underflow, need 2881 bytes.
E (7062) decoder: Buffer underflow, need 2881 bytes.
E (7072) decoder: Buffer underflow, need 2881 bytes.
E (7072) decoder: Buffer underflow, need 2881 bytes.
E (7082) decoder: Buffer underflow, need 2881 bytes.
E (7082) decoder: Buffer underflow, need 2881 bytes.
E (7092) decoder: Buffer underflow, need 2881 bytes.
E (7102) decoder: Buffer underflow, need 2881 bytes.
E (7102) decoder: Buffer underflow, need 2881 bytes.
E (7112) decoder: Buffer underflow, need 2881 bytes.
E (7112) decoder: Buffer underflow, need 2881 bytes.
E (7122) decoder: Buffer underflow, need 2881 bytes.
E (7122) decoder: Buffer underflow, need 2881 bytes.
E (7132) decoder: Buffer underflow, need 2881 bytes.
E (7142) decoder: Buffer underflow, need 2881 bytes.
E (7142) decoder: Buffer underflow, need 2881 bytes.
E (7152) decoder: Buffer underflow, need 2881 bytes.
E (7152) decoder: Buffer underflow, need 2881 bytes.
E (7162) decoder: Buffer underflow, need 2881 bytes.
E (7162) decoder: Buffer underflow, need 2881 bytes.
E (7172) decoder: Buffer underflow, need 2881 bytes.
E (7182) decoder: Buffer underflow, need 2881 bytes.
E (7182) decoder: Buffer underflow, need 2881 bytes.
E (7192) decoder: Buffer underflow, need 2881 bytes.
E (7192) decoder: Buffer underflow, need 2881 bytes.
E (7202) decoder: Buffer underflow, need 2881 bytes.
E (7202) decoder: Buffer underflow, need 2881 bytes.
E (7212) decoder: Buffer underflow, need 2881 bytes.
E (7212) decoder: Buffer underflow, need 2881 bytes.
E (7222) decoder: Buffer underflow, need 2881 bytes.
E (7232) decoder: Buffer underflow, need 2881 bytes.
E (7232) decoder: Buffer underflow, need 2881 bytes.
E (7242) decoder: Buffer underflow, need 2881 bytes.
E (7242) decoder: Buffer underflow, need 2881 bytes.
E (7252) decoder: Buffer underflow, need 2881 bytes.
E (7252) decoder: Buffer underflow, need 2881 bytes.
E (7262) decoder: Buffer underflow, need 2881 bytes.
E (7272) decoder: Buffer underflow, need 2881 bytes.
E (7272) decoder: Buffer underflow, need 2881 bytes.
E (7282) decoder: Buffer underflow, need 2881 bytes.
E (7282) decoder: Buffer underflow, need 2881 bytes.
E (7292) decoder: Buffer underflow, need 2881 bytes.
E (7292) decoder: Buffer underflow, need 2881 bytes.
E (7302) decoder: Buffer underflow, need 2881 bytes.
E (7302) decoder: Buffer underflow, need 2881 bytes.
E (7312) decoder: Buffer underflow, need 2881 bytes.
E (7322) decoder: Buffer underflow, need 2881 bytes.
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0104 (reserved sample frequency value)
dec err 0x0101 (lost synchronization)
dec err 0x0201 (CRC check failed)
dec err 0x0101 (lost synchronization)
dec err 0x0101 (lost synchronization)
dec err 0x0102 (reserved header layer value)
dec err 0x0101 (lost synchronization)
Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x8011e147  A1      : 0x3ffd7490
A2      : 0x3ffd7554  A3      : 0x3ffc35d8  A4      : 0x3ffda9a4  A5      : 0x0000007f
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8015cea4  A9      : 0x3f43d9a8
A10     : 0x3ffc35d8  A11     : 0x3ffd7554  A12     : 0x00000050  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x0000045a  SAR     : 0x00000018  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x00000000:0x3ffd7490 0x4011e147:0x3ffd74c0

Rebooting...
ets Jun  8 2016 00:22:57
```
…

aifer2007, 2017-05-15:
cg.mp3 info:

bit rate: 192 kbps
sample rate: 44.100 kHz
id3 : v2.3
code format : Lavf56.4.101

MrBuddyCasino, 2017-05-15:
> May I send the mp3 to you?

Sure, just upload it to Google Drive or something. You can send the link to buschfunk@yahoo.com if you don't want to write it here.

aifer2007, 2017-05-16:
Hi, @MrBuddyCasino ,

I have sent the mp3 file  to buschfunk@yahoo.com.

aifer2007, 2017-05-16:
I try much aac and m4a , but always get errors like this:

```
E (7062) audio_player: unknown mime type: 0
E (7062) audio_player: failed to start decoder task
```

MrBuddyCasino, 2017-05-16:
Your webserver might be misconfigured - you need to send the proper mime type header.
audio/aac => .aac
audio/mp4 => .mp4
audio/x-m4a => .m4a
audio/mpeg => .mp3

aifer2007, 2017-05-16:
Oh, I’ll try.

pablomorales88, 2017-05-30:
Hi, I've problem with underflow buffer when I listen music via URL. 
The Error is this type "buffer underflow".
I think the problem is the internet conection. But, i think that 10Mb of bandwidht the internet is sufficient for streaming radio. 

E (7132) decoder: Buffer underflow, need 2881 bytes.
E (7142) decoder: Buffer underflow, need 2881 bytes.
E (7142) decoder: Buffer underflow, need 2881 bytes.
E (7152) decoder: Buffer underflow, need 2881 bytes.
E (7152) decoder: Buffer underflow, need 2881 bytes.
E (7162) decoder: Buffer underflow, need 2881 bytes.
E (7162) decoder: Buffer underflow, need 2881 bytes.
E (7172) decoder: Buffer underflow, need 2881 bytes.
E (7182) decoder: Buffer underflow, need 2881 bytes.
E (7182) decoder: Buffer underflow, need 2881 bytes.
E (7192) decoder: Buffer underflow, need 2881 bytes.
E (7192) decoder: Buffer underflow, need 2881 bytes.
E (7202) decoder: Buffer underflow, need 2881 bytes.
E (7202) decoder: Buffer underflow, need 2881 bytes.
E (7212) decoder: Buffer underflow, need 2881 bytes.

Best regards.

MrBuddyCasino, 2017-05-31:
@pablomorales88 you can try increasing the buffer via #define SPIRAMSIZE to maybe 64k in spiram_fifo.c. Other than that, you might have crappy wifi or a particular problematic station, try to experiment. 
Also please open a separate issue next time, this is getting confusing.

pablomorales88, 2017-06-02:
Thank you for response, but, I still have the same problem and now I am trying to. Best Regards.

donny681, 2017-07-26:
`I (27229) mad_decoder: decoder start
I (27929) audio_player: Buffer fill 95%, 30699 bytes
I (28879) audio_player: Buffer fill 95%, 30686 bytes
E (31319) mad_decoder: Buffer underflow, need 2881 bytes.
E (31319) mad_decoder: Buffer underflow, need 2881 bytes.
E (31319) mad_decoder: Buffer underflow, need 2881 bytes.
E (31329) mad_decoder: Buffer underflow, need 2881 bytes.
E (31329) mad_decoder: Buffer underflow, need 2881 bytes.
E (31339) mad_decoder: Buffer underflow, need 2881 bytes.
E (31349) mad_decoder: Buffer underflow, need 2881 bytes.
E (31349) mad_decoder: Buffer underflow, need 2881 bytes.
E (31359) mad_decoder: Buffer underflow, need 2881 bytes.
E (31359) mad_decoder: Buffer underflow, need 2881 bytes.
E (31369) mad_decoder: Buffer underflow, need 2881 bytes.
E (31379) mad_decoder: Buffer underflow, need 2881 bytes.
E (31379) mad_decoder: Buffer underflow, need 2881 bytes.
E (31389) mad_decoder: Buffer underflow, need 2881 bytes.
E (31389) mad_decoder: Buffer underflow, need 2881 bytes.
E (31399) mad_decoder: Buffer underflow, need 2881 bytes.
E (31409) mad_decoder: Buffer underflow, need 2881 bytes.
E (31409) mad_decoder: Buffer underflow, need 2881 bytes.
E (31419) mad_decoder: Buffer underflow, need 2881 bytes.
E (31419) mad_decoder: Buffer underflow, need 2881 bytes.
E (31429) mad_decoder: Buffer underflow, need 2881 bytes.
E (31439) mad_decoder: Buffer underflow, need 2881 bytes.
E (31439) mad_decoder: Buffer underflow, need 2881 bytes.
E (31449) mad_decoder: Buffer underflow, need 2881 bytes.
E (31459) mad_decoder: Buffer underflow, need 2881 bytes.
E (31459) mad_decoder: Buffer underflow, need 2881 bytes.
E (31469) mad_decoder: Buffer underflow, need 2881 bytes.
`
I met the same problem.Maybe the reason is that the network is not well.I suggest that adding a xTaskDelay() in MP3_decoder.c:Line 79  if the buffer underflow.Or the task can't feed the watchdog and the chip will restart.After a  few seconds it may be work when showing this err.

`  ESP_LOGE(TAG, "Buffer underflow, need %d bytes.", buf_free_capacity_after_purge(buf));
            buf_underrun_cnt++;
            //We both silence the output as well as wait a while by pushing silent samples into the i2s system.
            //This waits for about 200mS
            renderer_zero_dma_buffer();
            vTaskDelay(1 / portTICK_PERIOD_MS);`

MrBuddyCasino, 2017-07-26:
Waiting until the buffer is x % full before resuming playback (instead of immediately) might fix it.

68a, 2017-10-31:
vTaskDelay(500/ portTICK_PERIOD_MS);

That is better.

Primus007, 2018-05-22:
i have the same problem as donny681. Buffer underflow

where must i write the "vTaskDelay(500/ portTICK_PERIOD_MS);" ?? or there are other ways to fix it?

Primus007, 2018-05-23:
Here is my Log:
`[0m
[0;32mI (5841) http_client: ... socket send success[0m
[0;32mI (6861) audio_player: Buffer fill 25%, 16026 bytes[0m
[0;32mI (7781) audio_player: Buffer fill 49%, 31998 bytes[0m
[0;32mI (8611) audio_player: Buffer fill 75%, 48009 bytes[0m
[0;32mI (9521) audio_player: RAM left 143352[0m
[0;32mI (9521) audio_player: created decoder task: mp3_decoder_task[0m
[0;32mI (9521) mad_decoder: decoder start[0m 
[0;31mE (9531) mad_decoder: dec err 0x0235 (bad main_data_begin pointer)[0m
[0;31mE (9531) mad_decoder: dec err 0x0235 (bad main_data_begin pointer)[0m
[0;32mI (9841) audio_player: Buffer fill 79%, 51061 bytes[0m
[0;32mI (11061) audio_player: Buffer fill 42%, 26896 bytes[0m
[0;32mI (12291) audio_player: Buffer fill 8%, 5264 bytes[0m
[0;31mE (12591) mad_decoder: Buffer underflow, need 2572 bytes.[0m
[0;31mE (12591) mad_decoder: Buffer underflow, need 2572 bytes.[0m
[0;31mE (12591) mad_decoder: Buffer underflow, need 2572 bytes.[0m
[0;31mE (12741) mad_decoder: Buffer underflow, need 2861 bytes.[0m
[0;31mE (12741) mad_decoder: Buffer underflow, need 2861 bytes.[0m
[0;31mE (12741) mad_decoder: Buffer underflow, need 2861 bytes.[0m
[0;31mE (12741) mad_decoder: Buffer underflow, need 2861 bytes.[0m
[0;31mE (12751) mad_decoder: Buffer underflow, need 2861 bytes.[0m`

Please can help me anyone??

danxster, 2018-06-17:
I'm also facing the same Buffer underflow issue. 

"vTaskDelay(500/ portTICK_PERIOD_MS);" made no difference.

I can confirm the esa-n fork by Yu Xutian works flawlessly:
https://github.com/esa-n/ESP32_MP3_Decoder.git

BeancurdPachelbel, 2018-10-03:
Yes, you are right @danxster

MrBuddyCasino, 2018-10-03:
It gets a lot better when AAC support is left out, because more RAM is available to buffering. This could be optimized to work much better for MP3 decoding. Not sure if I have the time for that currently.