MrBuddyCasino/ESP32_MP3_Decoder #9: where is GetAacDecoder() lib function?

aifer2007, 2017-05-21:
In aacdecoder.cpp in fdk-aac lib, 

CAacDecoder_Open() call GetAacDecoder() function.

GetAacDecoder() is not declared and defined, but can execute well.

Where is GetAacDecoder() declared and defined?

MrBuddyCasino, 2017-05-21:
Sorry, I have no idea. Please use a site like Stackoverflow for generic question, this is for project-related issues. I don't know many of the details of 3rd party code.

closed 2017-05-21 by MrBuddyCasino

aifer2007, 2017-05-21:
Thank you. 

ESP32_MP3_Decoder uses fdk-aac lib to decode AACs.
In ESP32_MP3_Decoder/components/fdk-aac/libAACdec/src/aacdecoder.cpp, in CAacDecoder_Open(), GetAacDecoder() is called to get handle of aacdecoder, but no codes declare or define it in all the respo. Yet it works well, no compiler or linker error.

Strange!

aifer2007, 2017-05-23:
It is defined via a macro, pls see:

https://github.com/mstorsjo/fdk-aac/issues/63#issuecomment-303018680