MrBuddyCasino/ESP32_MP3_Decoder #13: Fail to advance to the next track

donny681, 2017-06-24:
When pressing the IO0 in the mode of  the web radio,the chip will reboot.

` (6564) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (6574) main: RAM left 78588
I (6584) http_client: DNS lookup succeeded. IP=113.107.191.21
I (6584) http_client: ... allocated socket
I (9234) http_client: ... connected
I (9234) http_client: requesting GET /resource/n1/2009/12/31/655859227.mp3 HTTP/1.0
Host: win.web.ra01.sycdn.kuwo.cn

I (9244) http_client: ... socket send success
I (9354) audio_player: Buffer fill 33%, 15851 bytes
I (9444) audio_player: Buffer fill 65%, 31508 bytes
I (9494) audio_player: RAM left 80764
I (9494) audio_player: created decoder task: mp3_decoder_task
I (9494) mad_decoder: decoder start
I (9514) audio_player: Buffer fill 92%, 44270 bytes
I (10254) audio_player: Buffer fill 99%, 47764 bytes
I (11344) audio_player: Buffer fill 96%, 46337 bytes
I (12284) audio_player: Buffer fill 98%, 47416 bytes
I (13384) audio_player: Buffer fill 95%, 45948 bytes
I (14324) audio_player: Buffer fill 97%, 47029 bytes
I (15264) wifi: pm start, type:0

I (15414) audio_player: Buffer fill 94%, 45543 bytes
I (16364) audio_player: Buffer fill 97%, 46624 bytes
I (17304) audio_player: Buffer fill 99%, 47661 bytes
I (18394) audio_player: Buffer fill 96%, 46234 bytes
I (19344) audio_player: Buffer fill 98%, 47259 bytes
I (20434) audio_player: Buffer fill 95%, 45833 bytes
I (21374) audio_player: Buffer fill 97%, 46800 bytes
I (22324) audio_player: Buffer fill 99%, 47882 bytes
I (22814) web_radio: GPIO[0] intr, val: 1
I (22814) web_radio: RAM left 52092
W (22814) web_radio: next track: SomaFM / Secret Agent
I (22834) mad_decoder: decoder stopped
Untested FreeRTOS function uxTaskGetStackHighWaterMark
/Users/sky/esp/esp-idf/components/freertos/./tasks.c:3752 (uxTaskGetStackHighWaterMark)- assert failed!
abort() was called at PC 0x400853d2 on core 1
0x400853d2: uxTaskGetStackHighWaterMark at /Users/sky/esp/esp-idf/components/freertos/./tasks.c:3442

Backtrace: 0x40087688:0x3ffeed20 0x40087787:0x3ffeed40 0x400853d2:0x3ffeed60 0x4011e886:0x3ffeed80
0x40087688: invoke_abort at /Users/sky/esp/esp-idf/components/esp32/./panic.c:519

0x40087787: abort at /Users/sky/esp/esp-idf/components/esp32/./panic.c:519

0x400853d2: uxTaskGetStackHighWaterMark at /Users/sky/esp/esp-idf/components/freertos/./tasks.c:3442

0x4011e886: mp3_decoder_task at /Users/sky/esp/esp_demo/ESP32_MP3_Decoder/components/mp3_decoder/./mp3_decoder.c:181

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4404
load:0x40078000,len:11080
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

I (47) boot: ESP-IDF v2.0-rc1-1089-g3bd3640d-dirty 2nd stage bootloader
I (48) boot: compile time 17:23:18
I (60) boot: Enabling RNG early entropy source...
I (110) boot: SPI Speed      : 40MHz
I (149) boot: SPI Mode       : DIO
I (187) boot: SPI Flash Size : 4MB
I (225) boot: Partition Table:
I (259) boot: ## Label            Usage          Type ST Offset   Length
I (327) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (397) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (466) boot:  2 factory          factory app      00 00 00010000 00100000
I (536) boot: End of partition table
I (575) boot: Disabling RNG early entropy source...
I (627) boot: Loading app partition at offset 00010000
I (3130) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (3131) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x49b08 (301832) map
I (3181) boot: segment 2: paddr=0x00069b18 vaddr=0x3ffc0000 size=0x04ad4 ( 19156) load
I (3283) boot: segment 3: paddr=0x0006e5f4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (3341) boot: segment 4: paddr=0x0006e9fc vaddr=0x40080400 size=0x13bc0 ( 80832) load
I (3518) boot: segment 5: paddr=0x000825c4 vaddr=0x400c0000 size=0x00000 (     0) load
I (3519) boot: segment 6: paddr=0x000825cc vaddr=0x00000000 size=0x0da3c ( 55868) 
I (3575) boot: segment 7: paddr=0x00090010 vaddr=0x400d0018 size=0x976fc (620284) map
0x400d0018: _stext at ??:?

I (3654) cpu_start: Pro cpu up.
I (3688) cpu_start: Starting app cpu, entry point is 0x40080f08
0x40080f08: call_start_cpu1 at /Users/sky/esp/esp-idf/components/esp32/./cpu_start.c:186

I (3749) cpu_start: App cpu up.
I (3787) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (3856) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (3918) heap_alloc_caps: At 3FFD73D0 len 00008C30 (35 KiB): DRAM
I (3981) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (4045) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (4110) heap_alloc_caps: At 40093FC0 len 0000C040 (48 KiB): IRAM
I (4172) cpu_start: Pro cpu start user code
I (4332) cpu_start: Starting scheduler on PRO CPU.
I (4334) cpu_start: Starting scheduler on APP CPU.
I (4334) main: starting app_main()
I (4334) main: RAM left: 148020
I (4354) main: hardware initialized
I (4354) main: starting network
I (4364) wifi: wifi firmware version: bf86460
I (4364) wifi: config NVS flash: enabled
I (4364) wifi: config nano formating: disabled
I (4364) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (4374) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (4404) wifi: Init dynamic tx buffer num: 32
I (4404) wifi: Init dynamic rx buffer num: 32
I (4404) wifi: wifi driver task: 3ffdfb08, prio:23, stack:4096
I (4404) wifi: Init static rx buffer num: 10
I (4404) wifi: Init dynamic rx buffer num: 32
I (4414) wifi: Init rx ampdu len mblock:7
I (4414) wifi: Init lldesc rx ampdu entry mblock:4
I (4424) wifi: wifi power manager task: 0x3ffe56c4 prio: 21 stack: 2560
I (4424) wifi: wifi timer task: 3ffe672c, prio:22, stack:3584
I (4454) phy: phy_version: 355.1, 59464c5, Jun 14 2017, 20:25:06, 0, 0
I (4454) wifi: mode : sta (24:0a:c4:05:94:88)
I (4454) wifi: received wifi event 2
I (4464) wifi: connected successfully
I (4584) wifi: n:6 2, o:1 0, ap:255 255, sta:6 2, prof:1
I (5234) wifi: state: init -> auth (b0)
I (5244) wifi: state: auth -> assoc (0)
I (5264) wifi: state: assoc -> run (10)
I (5374) wifi: connected with TP-LINK_42AD, channel 6
I (5384) wifi: received wifi event 4
`

MrBuddyCasino, 2017-06-25:
Can you attach your .pls?

n24bass, 2017-06-28:
You'd unset configuration for "Halt when an SMP-untested function is called" in FreeRTOS.

MrBuddyCasino, 2017-06-28:
@n24bass: you're right, its right there in the stacktrace. *doh*
I'll add a proper default to the kconfig.

MrBuddyCasino, 2017-07-12:
Should be fixed now.

closed 2017-07-12 by MrBuddyCasino