MrBuddyCasino/ESP32_MP3_Decoder #14: PDM fails to play after half an audio data packet on bluetooth(p_data is null)

joeybab3, 2017-06-28:
After getting it to work(note: cmd is not gnd...), I was playing around with it and I can't get it to play over bluetooth when the audio mode is set to pdm. It gives the error below:
```
I (2549) main: RAM left 134032
E (21349) BT: btm_sec_conn_req

E (21369) BT: btm_sec_connected

I (21879) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0
I (21879) BT_AV: a2dp conn_state_cb, state 1
I (22369) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x2
I (22369) BT_AV: a2dp audio_cfg_cb , codec type 0
I (22369) BT_AV: audio player configured
E (22899) BT: bta_dm_rm_cback:0, status:0
I (22899) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0
I (22899) BT_AV: a2dp conn_state_cb, state 2
E (23809) BT: process_service_search_attr_rsp

E (23959) BT: bta_av_rc_opened allow incoming AVRCP connections:1
E (23959) BT: bta_av_rc_opened rcb[0] shdl:1 lidx:1/0
I (23969) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0
I (23969) BT_AV: avrc conn_state evt: state 1, feature 0x259, [9c:2a:83:2c:d5:d6
]
E (29339) BT: bta_av_proc_stream_evt: p_data is null
E (29339) BT: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:
x8202

E (29339) BT: bta_dm_rm_cback:0, status:0
E (29339) BT: bta_dm_rm_cback:0, status:7
E (29349) BT: BTA_AV_START_EVT status 0, suspending 0, init 0

I (29349) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x1
I (29359) BT_AV: a2dp audio_state_cb state 2
E (31539) BT_AV: audio data pkt cnt 100
```
and just sounds like it repeats the same sample over and over.
It doesn't have this issue when streaming over wifi.

MrBuddyCasino, 2017-06-28:
Thanks for the report, I'll look into it as soon as I find the time.

fanthos, 2017-09-11:
I got the same error after pkt count 700 printed, and have to reset the esp32 to get bluetooth works again.