MrBuddyCasino/ESP32_MP3_Decoder #15: another mp3 stream

zajaklis, 2017-07-13:
i was trying to play http://lr1mp1.latvijasradio.lv:8012 but it seems it requires "Icy-MetaData:1" header in GET. Stream plays ok in vlc and another players.
Codec: MPEG Audio layer 1/2 (mpga)
Sample rate: 44100 Hz
Bitrate: 128 kb/s
content type is not returned as audio/mpeg type so there must be another way to catch stream

MrBuddyCasino, 2017-07-14:
I don't think mp2  is supported. Headers are currently mandatory, maybe we could add a default to assume mp3, or parse the first few bytes? Pull requests welcome.

MrBuddyCasino, 2017-07-14:
Well whaddaya know, someone made a pull request to add MP2 support. I just merged it.

spockbear, 2017-07-14:
playlist.c:19:17: fatal error: ini.h: No such file or directory
What is the problem ???

zajaklis, 2017-07-14:
spockbear, have you read readme? `git submodule init && git submodule update`

MrBuddyCasino, 2017-07-15:
@zajaklis does the new version solve your issue?

zajaklis, 2017-07-15:
patch for mpeg 1 and 2 did not resolved issue as my provided radio link seems to be mp3 format:
just checked:

> http://lr1mp1.latvijasradio.lv:8012 or http://lr1mp1.latvijasradio.lv:8012/listen.pls (does not work)
> Audio: MP3 44100Hz stereo 96kbps [A: mp3, 44100 Hz, stereo, 96 kb/s]

> http://skonto.ls.lv:8002/mp3 (plays ok)
> Audio: MP3 44100Hz stereo 128kbps [A: mp3, 44100 Hz, stereo, 128 kb/s]

format for both seems to be same as reported by Media Player Classic
more info for captured audio from http://lr1mp1.latvijasradio.lv:8012:

```
ICY 200 OK
icy-notice1:<BR>This stream requires <a href="http://www.winamp.com/">Winamp</a><BR>
icy-notice2:SHOUTcast Distributed Network Audio Server/win32 v1.9.8<BR>
icy-name:Radio 1
icy-genre:classical
icy-url:http://lr1.latvijasradio.lv
content-type:audio/mpeg
icy-pub:0
icy-br:128
```

MrBuddyCasino, 2017-07-15:
Ok I thought it might because you said codec was MPEG Audio layer 1/2 (mpga). I'll investigate.

zajaklis, 2017-07-15:
well, MPEG Audio layer 1/2 (mpga) was reported by VLC for both of my mentioned streams (even for the one which works ok) so i tried another player just to make sure its reports mp3 stream.

MrBuddyCasino, 2017-07-15:
Sorry, can't find the reason why it doesn't work.
This is cURL:

````
curl -v --http1.0 http://lr1mp1.latvijasradio.lv:8012/
*   Trying 159.148.56.131...
* TCP_NODELAY set
* Connected to lr1mp1.latvijasradio.lv (159.148.56.131) port 8012 (#0)
> GET / HTTP/1.0
> Host: lr1mp1.latvijasradio.lv:8012
> User-Agent: curl/7.51.0
> Accept: */*
>
ICY 200 OK
icy-notice1:<BR>This stream requires <a href="http://www.winamp.com/">Winamp</a><BR>
icy-notice2:SHOUTcast Distributed Network Audio Server/win32 v1.9.8<BR>
icy-name:Radio 1
icy-genre:classical
icy-url:http://lr1.latvijasradio.lv
content-type:audio/mpeg
icy-pub:0
icy-br:128
````

This is the ESP:
````
I (3873) http_client: requesting GET / HTTP/1.0
Host: lr1mp1.latvijasradio.lv:8012
User-Agent: curl/7.51.0
Accept: */*
````

cURL works, ESP doesn't. No idea why. You can try and contact the server operator to have a look at requests with user-agents of "ESP32" and see whats going on in the logs.

zajaklis, 2017-07-17:
got it working,
http_parser.c expects `HTTP/1.0 200 OK` in http response header
but some web stream services sends `ICY 200 OK`
so idea was to add ICY response support to http parser:
`  , s_res_I
  , s_res_IC
`

MrBuddyCasino, 2017-07-18:
ICY 200 OK. Dear lord, why.
Thanks.

shaurya0406, 2018-02-28:
I am having the same problem:
playlist.c:19:17: fatal error: ini.h: No such file or directory
 I have tried   git submodule init && git submodule update
but to no avail.
please help
thanks in avance
Regards

shaurya0406, 2018-02-28:
same issue as spockbear

any help??

MrBuddyCasino, 2018-02-28:
Does this work?
```
git clone https://github.com/MrBuddyCasino/ESP32_MP3_Decoder.git
cd ESP32_MP3_Decoder/
git submodule init && git submodule update
make -j4 flash
```

shaurya0406, 2018-02-28:
YES it worked!!!
THANK YOU VERY MUCH
Regards

closed 2018-03-02 by MrBuddyCasino