MrBuddyCasino/ESP32_MP3_Decoder #19: Link error with the audio decoder demo.

13824125580, 2017-07-18:
Hi Michael:

  i encounter a link error when i use your web audio decoder demo with esp32, the error report is as blow, and i am puzzle, dont know why,  can you give me some hints？　appreciate for you help.

LD mp3_decoder.elf
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/build/http/libhttp.a(http.o):(.literal.http_client_get+0x48): undefined reference to `**http_parser_init**'
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/build/http/libhttp.a(http.o):(.literal.http_client_get+0x4c): undefined reference to `**http_parser_execute**'
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/build/http/libhttp.a(http.o): In function `http_client_get':
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/components/http/./http.c:93: undefined reference to `http_parser_init'
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/components/http/./http.c:105: undefined reference to `http_parser_execute'
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/build/url_parser/liburl_parser.a(url_parser.o):(.literal.url_create+0x10): undefined reference to `http_parser_parse_url'
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/build/url_parser/liburl_parser.a(url_parser.o): In function `url_create':
/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/components/url_parser/./url_parser.c:113: undefined reference to `http_parser_parse_url'
collect2: error: ld returned 1 exit status
make: *** [/home/zilong/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio/build/mp3_decoder.elf] 错误 1
zilong@PCZiLong:~/WorkSpace/IOT/esp32/esp32audio/esp32-projects/web_radio$

MrBuddyCasino, 2017-07-18:
Did you do `git submodule init && git submodule update`?

13824125580, 2017-07-19:
i just do the git clone --recursive frome the beginning,  i will try your suggestion and feedback, thank you.

13824125580, 2017-07-19:
not work :(

MrBuddyCasino, 2017-07-19:
Just tested again on my machine, works. Do you have the latest ESP-IDF? We're on the master branch from Github.

MrBuddyCasino, 2017-08-22:
Closing due to inactivity.

closed 2017-08-22 by MrBuddyCasino

rock-dong, 2017-10-12:
I reproduce the problem, it is caused by the esp-idf code release.  In the GitHub, esp deliver a bug fix at July 26 "nghttp: build port directory " . Please refer to 
https://github.com/espressif/esp-idf/commit/ee6d0631f4c2a6a512cbc3f043c70d45ff7ab68d#diff-df69c0ddcfe8e93525541d26f1050230