MrBuddyCasino/ESP32_MP3_Decoder #21: Can not run the demo by the wifi radio

donny681, 2017-07-24:
`MONITOR
--- WARNING: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.SLAB_USBtoUART instead...
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:5084
load:0x40078000,len:0
load:0x40078000,len:12652
entry 0x40078f44
I (45) boot: ESP-IDF v3.0-dev-174-g143cdf6a-dirty 2nd stage bootloader
I (45) boot: compile time 10:54:45
I (48) boot: Enabling RNG early entropy source...
I (64) boot: SPI Speed      : 40MHz
I (77) boot: SPI Mode       : DIO
I (89) boot: SPI Flash Size : 4MB
I (102) boot: Partition Table:
I (113) boot: ## Label            Usage          Type ST Offset   Length
I (136) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (159) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (182) boot:  2 factory          factory app      00 00 00010000 00100000
I (206) boot: End of partition table
I (219) boot: Disabling RNG early entropy source...
I (236) boot: Loading app partition at offset 00010000
I (254) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4d2d4 (316116) map
I (618) esp_image: segment 1: paddr=0x0005d2fc vaddr=0x3ffb0000 size=0x02d14 ( 11540) load
I (633) esp_image: segment 2: paddr=0x00060018 vaddr=0x400d0018 size=0x9a55c (632156) map
0x400d0018: _stext at ??:?

I (1312) esp_image: segment 3: paddr=0x000fa57c vaddr=0x3ffb2d14 size=0x020e4 (  8420) load
I (1323) esp_image: segment 4: paddr=0x000fc668 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (1335) esp_image: segment 5: paddr=0x000fca70 vaddr=0x40080400 size=0x13f5c ( 81756) load
I (1465) esp_image: segment 6: paddr=0x001109d4 vaddr=0x400c0000 size=0x00000 (     0) load
E (1466) esp_image: Image length 1051136 doesn't fit in partition length 1048576
E (1483) boot: Failed to verify app image @ 0x10000 (8194)
user code done`

I update the latest version and find that there is something wrong.It shows that image overflows.

MrBuddyCasino, 2017-07-24:
Compiles fine for me with the latest SDK. Did you set any particular flash size?

MrBuddyCasino, 2017-07-24:
Ah it complains on startup. I added a custom partition table, but sdkconfig.defaults don't seem to work. You might have to select "custom partition" manually in menuconfig. Does this solve your issue?

donny681, 2017-07-26:
OK,it can work.But I found another problem.

` (78999) web_radio: RAM left 4468
I (79049) mad_decoder: decoder stopped
I (79049) mad_decoder: MAD decoder stack: 504

I (79069) http_client: DNS lookup succeeded. IP=121.14.255.95
I (79069) http_client: ... allocated socket
I (79169) http_client: ... connected
I (79179) http_client: requesting GET /79ca15e1408967ca59bb8d4eeda3a3c1/597835fe/G098/M08/07/19/AocBAFkn4PmASTuWADWju6kYBPk145.mp3 HTTP/1.0
Host: fs.web.kugou.com:80
User-Agent: ESP32
Accept: */*

I (79189) http_client: ... socket send success
I (79309) audio_player: Buffer fill 3%, 2079 bytes
I (79639) audio_player: Buffer fill 28%, 18051 bytes
I (79769) audio_player: Buffer fill 53%, 34036 bytes
I (79839) audio_player: Buffer fill 78%, 50047 bytes
I (79899) audio_player: RAM left 27536
I (79899) audio_player: created decoder task: mp3_decoder_task
E (79899) common: couldn't allocate buffer of size 2889
E (79909) mad_decoder: buf_create() failed

Guru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x4012183e  PS      : 0x00060730  A0      : 0x00000000  A1      : 0x3fff6480  
0x4012183e: mp3_decoder_task at /Users/sky/esp/esp_demo/ESP32_MP3_Decoder/components/mp3_decoder/./mp3_decoder.c:124 (discriminator 2)

A2      : 0x3ffdc7c8  A3      : 0x3fffff98  A4      : 0x3fff6510  A5      : 0x3fffb8b0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8012183e  A9      : 0x3fff6430  
A10     : 0x4010c484  A11     : 0x3ffc7bcc  A12     : 0x3f410a20  A13     : 0x0000001f  
0x4010c484: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:35

A14     : 0x00000001  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x4012183e:0x3fff6480 0x7ffffffd:0x3fff64a0
0x4012183e: mp3_decoder_task at /Users/sky/esp/esp_demo/ESP32_MP3_Decoder/components/mp3_decoder/./mp3_decoder.c:124 (discriminator 2)`

If  playing the next song everytime,it will create several new task and malloc the new buffer.If not delete pre task and recycle the buffer which it allocate in time , it may consume the raw.At last the chip will restart.I'm trying to fix the bug.

MrBuddyCasino, 2017-07-26:
Looking forward to what you find out.

donny681, 2017-07-27:
**Eh,I found that Can not play the next song after finish playing the a song. I try to press the button.But it's no use.I check the header of the response.It can get the right response.But it can't read the data.Do you know the reason why it can't play the next song after finishing playing song.**
`I (2280) main: RAM left 177220
I (2300) http_client: DNS lookup succeeded. IP=121.12.98.117
I (2300) http_client: ... allocated socket
I (2310) http_client: ... connected
I (2310) http_client: requesting GET /fab801b84c8e65845f37dc1b1ae40a5d/59786b22/G057/M00/08/01/eQ0DAFbvc3CIW1AaAAmGmOHq2ZcAABDGQIbBRMACYaw853.mp3 HTTP/1.0
Host: fs.web.kugou.com:80
User-Agent: ESP32
Accept: */*

I (2330) http_client: ... socket send success

thu, 27 jul 2017 02:29:54 gmt
content-l

624280
accept-ranges: bytes
content-type: audio/mpeg

bytes
content-type: audio/mpeg

audio/mpeg

mon, 21 mar 2016 04:07:12 gmt
connection: close

close

Fw-Cache-Status: hit
Fw-Via: DISK HIT from 222.186.129.169, 

hit
fw-via: disk hit from 222.186.129.169, 

disk hit from 222.186.129.169, 

Configured MISS from 122.192.111.136, DISK HIT from 222.88.91.3

9, DISK HIT from 121.12.98.60

136, DISK HIT from 222.88.91.3I (2440) audio_player: Buffer fill 50%, 16076 bytes
I (2500) audio_player: RAM left 179324
I (2500) audio_player: created decoder task: mp3_decoder_task
I (2500) mad_decoder: decoder start
E (2510) mad_decoder: dec err 0x0101 (lost synchronization)
I (2520) audio_player: Buffer fill 91%, 29259 bytes
I (3410) audio_player: Buffer fill 94%, 30187 bytes
I (4360) audio_player: Buffer fill 97%, 31268 bytes
I (5450) audio_player: Buffer fill 93%, 29842 bytes
I (6400) audio_player: Buffer fill 96%, 30882 bytes
I (7340) audio_player: Buffer fill 99%, 31963 bytes
I (8430) audio_player: Buffer fill 95%, 30490 bytes
I (9380) audio_player: Buffer fill 98%, 31571 bytes
I (10470) audio_player: Buffer fill 94%, 30145 bytes
I (11410) audio_player: Buffer fill 97%, 31166 bytes
I (11460) wifi: pm start, type:0

I (12500) audio_player: Buffer fill 92%, 29740 bytes
I (13450) audio_player: Buffer fill 96%, 30775 bytes
I (14390) audio_player: Buffer fill 99%, 31826 bytes
I (15480) audio_player: Buffer fill 95%, 30400 bytes
I (16430) audio_player: Buffer fill 98%, 31458 bytes
I (17520) audio_player: Buffer fill 93%, 30032 bytes
I (18470) audio_player: Buffer fill 97%, 31066 bytes
I (19560) audio_player: Buffer fill 92%, 29640 bytes
I (20500) audio_player: Buffer fill 96%, 30722 bytes
I (21450) audio_player: Buffer fill 99%, 31752 bytes
I (22540) audio_player: Buffer fill 94%, 30326 bytes
I (23480) audio_player: Buffer fill 98%, 31360 bytes
[... 131 lines cut ...]

*
access-control-allow-headers: origin, accept, x-request

origin, accept, x-request

ed-With, Content-Type
Access-Control-Allow-Methods: GET, OPTIO

get, optio

NS, HEAD
Connection: Close
Expires: Mon, 26 Jul 1997 05:00:00

close
expires: mon, 26 jul 1997 05:00:00

mon, 26 jul 1997 05:00:00

 GMT

connection: close

`

MrBuddyCasino, 2017-07-27:
Never tested that case, I assumed infinite streaming.
Unfortunately my time is very limited right now.

reb810, 2017-09-13:
Hey @donny681, I'm having the same initial issue you had where the project flashes but gives the weird errors and 'user code done' in the monitor.  What exactly did you do to fix this issue?

Cheers