MrBuddyCasino/ESP32_MP3_Decoder #24: Need help about MP3 decode

z43740979, 2017-09-15:
Sir I can not run your demo 
ERROR：/ESP32_MP3_Decoder/components/playlist/./playlist.c:19:17: fatal error: ini.h: No such file or directory
And I want decode MP3 file in SD card but the stack overflow ，I'd appreciate for your help
Here is part of my code ：
static enum mad_flow input(void *data, struct mad_stream *stream)
{
    printf("input data \n");
    mp3_file *mp3fp;
    int ret_code;
    int unproc_data_size; /*the unprocessed data's size*/
    int copy_size;
    mp3fp = (mp3_file *)data;
    if(mp3fp->fpos < mp3fp->flen){
        unproc_data_size = stream->bufend - stream->next_frame;
        memcpy(mp3fp->fbuf, mp3fp->fbuf + mp3fp->fbsize - unproc_data_size, unproc_data_size);
        copy_size = BUFSIZE - unproc_data_size;
        if(mp3fp->fpos + copy_size > mp3fp->flen){
            copy_size = mp3fp->flen - mp3fp->fpos;
        }
        printf("copy_size=%d \n",copy_size);
        fread(mp3fp->fbuf+unproc_data_size, 1, copy_size, mp3fp->fp);
        mp3fp->fbsize = unproc_data_size + copy_size;
        mp3fp->fpos += copy_size;
        mad_stream_buffer(stream, mp3fp->fbuf, mp3fp->fbsize);
        ret_code = MAD_FLOW_CONTINUE;
    }else{
        ret_code = MAD_FLOW_STOP;
    }
    return (mad_flow)ret_code;
}

static int decode(mp3_file *mp3fp)
{
    struct mad_decoder decoder;
    int result;
    /* configure input, output, and error functions */
    mad_decoder_init(&decoder, mp3fp,
            input, 0 /* header */, 0 /* filter */, output,
            error, 0 /* message */);

    /* start decoding */
    result = mad_decoder_run(&decoder, MAD_DECODER_MODE_SYNC);

    /* release the decoder */
    mad_decoder_finish(&decoder);

    return result;
}

MrBuddyCasino, 2017-09-15:
Did you do a `git submodule init && git submodule update`?

z43740979, 2017-09-18:
mad_decoder: Buffer underflow, need 2530 bytes.
Is that means Not enough storage ？

MrBuddyCasino, 2017-09-18:
No, it means the data did not come in fast enough to sustain real-time audio playback. It happens sometimes. Try to improve your wifi connection, or try a different streaming URL.

Mscichu, 2017-11-11:
Hi!
MrBuddyCasino your code works great! Thanks!
But I'm try also decode MP3 from SD card. z43740979  did you successfully play MP3 from SD card?

closed 2018-06-29 by MrBuddyCasino