MrBuddyCasino/ESP32_MP3_Decoder #29: APLL doesn't work in DAC and PMD modes

AlexanderKoro, 2017-11-07:
WebRadio doesn't play in DAC and PDM modes when apll_use=1 (in PDM mode it is very slow, in DAC mode - some cracks). When set apll_use=0 radio works. Nothing changed except apll_use.

**Output from I2S PDM APLL=1 (not work):**
I (3756) renderer: init I2S mode 3, port 0, 16 bit, 44100 Hz
I (3756) renderer: chip revision 1, enabling APLL
...
I (3776) I2S: APLL: Req RATE: 44100, real rate: 44099.020, BITS: 16, CLKM: 1, BC K: 8, MCLK: 5.669, SCLK: 1411168.625000, diva: 1, divb: 0

**Output from I2S PDM APLL=0 (sound a little bit slower than needed which lead to server stop the connection after some time):**
I (3296) I2S: PLL_D2: Req RATE: 44100, real rate: 200000.000, BITS: 16, CLKM: 5, BCK: 5, MCLK: 5.669, SCLK: 6400000.000000, diva: 64, divb: 42

**Output from I2S DAC APLL=1 (not work):**
I2S: APLL: Req RATE: 44100, real rate: 44099.020, BITS: 16, CLKM: 1, BCK: 8, MCLK: 30.234, SCLK: 1411168.625000, diva: 1, divb: 0

**Output from I2S DAC APLL=0 (radio works for a long time, sound not clear):**
I (4286) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30, BCK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14

MrBuddyCasino, 2017-11-08:
Opened: https://github.com/espressif/esp-idf/issues/1244
Reply will determine further actions.

lacyrhoades, 2017-11-19:
Playing with apll_use=0 as well to use DAC as line level output – wondering though, what is APLL?

Assuming "[something] phase locked loop"

Only reference I could find is here: https://github.com/espressif/esp-idf/blob/master/docs/api-reference/peripherals/i2s.rst

MrBuddyCasino, 2017-11-19:
Its the Audio PLL, it improves noise noticeably.

MrBuddyCasino, 2018-03-02:
Fixed.

closed 2018-03-02 by MrBuddyCasino