MrBuddyCasino/ESP32_MP3_Decoder #32: ESP32 keeps crashing after streaming for 30 seconds

newkit, 2017-11-11:
I have the code running on a ESP-WROOM-32 with a [PCM5102A](https://www.ebay.de/itm/I2S-PCM5102A-DAC-Decoder-32bit-Player-Module-ES9023-PCM1794-For-Raspberry-Pi-New/272384686056?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) connected as follows:

```
ESP32        <------->    PCM5102A

GPIO 22                   DIN
GPIO 25                   LCK
GPIO 26                   BCK

FLT, DMP, SCL, FMT on the PCM5102A are on GND
XMT to VCC or other positive voltage, see below.
```
XMT hooked up to VCC as well I tried with a 4k/1k voltage diver to roughly 2.7V as I've seen in this [schematic](https://www.dropbox.com/s/q9ffrd3g8og1z5j/pidac_5102A.pdf?dl=0) 
Also tried powering the entire audio module with eighter 3.3V or 5V and adding a capacitor for voltage stabilisation, but with no effect in the errorneuos behavior.

Problem is, that after streaming fine for roughly 30s-2min, the ESP32 is crashing:

```
Tasks currently running:
CPU 0: BtcMediaT

CPU 1: IDLE
ASSERT_ERR((EM_BT_RXDESC_OFFSET + ld_env.curr_rxdesc_index * REG_EM_BT_RXDESC_SIZE) == bt_et_currentrxdescptr_currentrxdescptr_getf()), in ld_acl.c at line 1900ets Jun  8 2016 00:22:57
```

`ld_acl.c` seems to be part of the espressif BT implementation, but I couldn't the code to get an idea whats going wrong. I am pretty clueless what might be the issue. Any suggestions?
I am aware, that this issue is obviously not in the ESP32_MP3_Decoder code but I hope that someone here has an idea in what direction to look.

Any help is much appreciated! :-D

See the entire serial log here:
```
⸮)⸮⸮z⸮!`⸮⸮⸮
T!⸮9(⸮⸮⸮⸮ʄ⸮1	⸮ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5740
load:0x40078000,len:0
load:0x40078000,len:13760
entry 0x40079020
[0;32mI (69) boot: Detected ESP32[0m
[0;32mI (32) boot: ESP-IDF v3.0-dev-1123-g0ee9d93 2nd stage bootloader[0m
[0;32mI (32) boot: compile time 16:39:17[0m
[0;32mI (32) boot: Enabling RNG early entropy source...[0m
[0;32mI (38) boot: SPI Speed      : 40MHz[0m
[0;32mI (42) boot: SPI Mode       : DIO[0m
[0;32mI (46) boot: SPI Flash Size : 4MB[0m
[0;32mI (50) boot: Partition Table:[0m
[0;32mI (54) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (76) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (84) boot: End of partition table[0m
[0;32mI (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3e0cc (254156) map[0m
[0;32mI (185) esp_image: segment 1: paddr=0x0004e0f4 vaddr=0x3ffc0000 size=0x01f1c (  7964) load[0m
[0;32mI (188) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0x8eaf0 (584432) map[0m
[0;32mI (394) esp_image: segment 3: paddr=0x000deb10 vaddr=0x3ffc1f1c size=0x00b44 (  2884) load[0m
[0;32mI (396) esp_image: segment 4: paddr=0x000df65c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (402) esp_image: segment 5: paddr=0x000dfa64 vaddr=0x40080400 size=0x13574 ( 79220) load[0m
[0;32mI (443) esp_image: segment 6: paddr=0x000f2fe0 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (454) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (454) boot: Disabling RNG early entropy source...[0m
[0;32mI (455) cpu_start: Pro cpu up.[0m
[0;32mI (458) cpu_start: Starting app cpu, entry point is 0x40080fac[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (469) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (476) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (482) heap_init: At 3FFCFFF8 len 00010008 (64 KiB): DRAM[0m
[0;32mI (488) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (494) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (501) heap_init: At 40093974 len 0000C68C (49 KiB): IRAM[0m
[0;32mI (507) cpu_start: Pro cpu start user code[0m
[0;32mI (189) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (190) main: starting app_main()[0m
[0;32mI (190) main: RAM left: 173440[0m
[0;32mI (210) main: hardware initialized[0m
[0;32mI (210) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (510) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0[0m
[0;32mI (550) renderer: init I2S mode 0, port 0, 16 bit, 44100 Hz[0m
[0;32mI (550) I2S: queue free spaces: 1[0m
[0;32mI (550) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=32[0m
[0;32mI (560) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, divb: 11[0m
[0;32mI (570) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0[0m
[0;32mI (570) main: RAM left 131368[0m
[0;32mI (580) bt_speaker: starting "B00mB0X"[0m
[0;31mE (14220) BT: btm_sec_conn_req
[0m
[0;31mE (14240) BT: btm_sec_connected
[0m
[0;32mI (14500) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0[0m
[0;32mI (14500) BT_AV: a2dp conn_state_cb, state 1[0m
[0;32mI (14650) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x2[0m
[0;32mI (14650) BT_AV: a2dp audio_cfg_cb , codec type 0[0m
[0;32mI (14650) BT_AV: audio player configured[0m
[0;31mE (14710) BT: bta_dm_rm_cback:0, status:0[0m
[0;32mI (14720) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0[0m
[0;32mI (14720) BT_AV: a2dp conn_state_cb, state 2[0m
[0;31mE (14760) BT: process_service_search_attr_rsp
[0m
[0;31mE (14800) BT: bta_av_rc_opened allow incoming AVRCP connections:1[0m
[0;31mE (14800) BT: bta_av_rc_opened rcb[0] shdl:1 lidx:1/0[0m
[0;32mI (14800) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x0[0m
[0;32mI (14810) BT_AV: avrc conn_state evt: state 1, feature 0x249, [d8:a2:5e:80:bc:58][0m
[0;31mE (15080) BT: bta_av_proc_stream_evt: p_data is null[0m
[0;31mE (15080) BT: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x8202
[0m
[0;31mE (15080) BT: bta_dm_rm_cback:0, status:0[0m
[0;31mE (15090) BT: bta_dm_rm_cback:0, status:7[0m
[0;31mE (15100) BT: BTA_AV_START_EVT status 0, suspending 0, init 0
[0m
[0;32mI (15100) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x1[0m
[0;32mI (15110) BT_AV: a2dp audio_state_cb state 2[0m
[0;31mE (17800) BT_AV: audio data pkt cnt 100[0m
[0;31mE (20120) BT_AV: audio data pkt cnt 200[0m
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: BtcMediaT

CPU 1: IDLE
[0;31mE (22440) BT_AV: audio data pkt cnt 300[0m
[0;31mE (24760) BT_AV: audio data pkt cnt 400[0m
[0;31mE (27090) BT_AV: audio data pkt cnt 500[0m
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: BtcMediaT

CPU 1: IDLE
ASSERT_ERR((EM_BT_RXDESC_OFFSET + ld_env.curr_rxdesc_index * REG_EM_BT_RXDESC_SIZE) == bt_et_currentrxdescptr_currentrxdescptr_getf()), in ld_acl.c at line 1900
```

MrBuddyCasino, 2017-11-11:
Huh. I'm afraid I've never seen that one, is this with the latest ESP-IDF?

newkit, 2017-12-11:
Well, what ever it was, after checking it out again, and setting everything up, it now works :-D

closed 2017-12-11 by newkit

donny681, 2018-03-17:
@newkit I met the same question.How do you fix it?

donny681, 2018-03-17:
Do you close the watch dog? @newkit @MrBuddyCasino