MrBuddyCasino/ESP32_MP3_Decoder #36: Blue Tooth Enable

cr8myf8, 2017-12-10:
Very cool project!  

I was able to build the project and flash it onto the sparkfun esp32 thing board.  I can play an MP3 stream with the built in DAC, but I can't enable the bluetooth in the menuconfig.  Any idea why the bluetooth is not an option?

MrBuddyCasino, 2017-12-10:
You first have to enable the generic bluetooth support in the ESP32 config. Its somewhere in the component configuration menu I think.

cr8myf8, 2017-12-10:
Thanks for the response.  I am new to the esp32 so I am sorry if my questions seem ignorant.  I was able to flash the esp32 with the bluetooth a2dp_sink example code.  The config menu showed some bluetooth options.  However, the configuration menu for this project is just blank.  Maybe I messed up the files.  I will try and clone the project again to see if that fixes the issue.

Configuration Menu for ESP32_MP3_Decoder
[image]

Configuration Menu for a2dp_sink
[image]

MrBuddyCasino, 2017-12-11:
Hmm, looks like they changed some stuff - possibly a v3 problem. Not sure when I get the time to look into it, you can try editing the kconfig file, there is condition in there that disables bluetooth speaker mode unless esp32 bt is enabled. Maybe they renamed it.

MrBuddyCasino, 2018-01-17:
Updated code, please try again. Make sure to enable A2DP Sink in menuconfig.

closed 2018-06-29 by MrBuddyCasino