MrBuddyCasino/ESP32_MP3_Decoder #37: The speaker does not ring

LilyGO, 2017-12-11:
minicom shows the audio data in transit, but there is no volume. My din gain pin is not connected, and my module is max98357A

minicom:
E (94116) BT: BTA_AV_START_EVT status 0, suspending 0, init 0

I (94116) BT_APP_CORE: bt_app_task_handler, sig 0x1, 0x1
I (94126) BT_AV: a2dp audio_state_cb state 2
E (96146) BT_AV: audio data pkt cnt 100
E (98146) BT_AV: audio data pkt cnt 200
E (100146) BT_AV: audio data pkt cnt 300
E (102156) BT_AV: audio data pkt cnt 400
E (104166) BT_AV: audio data pkt cnt 500
E (106166) BT_AV: audio data pkt cnt 600
E (108166) BT_AV: audio data pkt cnt 700

MrBuddyCasino, 2017-12-11:
Did you configure I2S output?

LilyGO, 2017-12-11:
[image]

LilyGO, 2017-12-11:
I keep the I2S default settings, the rest there is a need to set the place?

MrBuddyCasino, 2017-12-11:
Looks good. I suppose you already checked the pin connections. Do you have an oscilloscope? The I2S signals are quite fast and need decent electrical connections and short wires to work.

LilyGO, 2017-12-12:
[image]

LilyGO, 2017-12-12:
Maybe my lead is really too long. thanks for your advice

closed 2018-02-14 by MrBuddyCasino