MrBuddyCasino/ESP32_MP3_Decoder #41: /mdns_task.c:29:5: error: unknown type name 'mdns_server_t'

vibnwis, 2018-01-24:
Experiencing the error and used the command grep -rnw '.' -e 'mdns_server_t' to find the definition but not found. Very strange! Any advice? please

MrBuddyCasino, 2018-01-24:
I removed that part with the latest commit, or so I thought?

vibnwis, 2018-01-24:
I am compiling the code from  kodera2t. Obviously, it still uses the old code. I do not how the repositories work when someone developed on someone old code but which has been updated recently. Any idea?

MrBuddyCasino, 2018-01-24:
Well then you're barking up the wrong tree. Open an issue with kodera2t's repository then or switch to my version.

vibnwis, 2018-01-24:
Understood. Apologies for being inconvenient at your end. However, it would be a great help If possible you can share a bit.  If someone create a new repo out of someone else (you for instance), and when you update your repo, will it get propagated to those persons' repo? or they can maintain an old repo as long as they want?

MrBuddyCasino, 2018-01-24:
It won't get automatically updated. They can pull the changes if they want to, but if they don't then the two repos will drift out of sync.

vibnwis, 2018-01-24:
Very appreciated. Cheers!

closed 2018-01-24 by MrBuddyCasino