MrBuddyCasino/ESP32_MP3_Decoder #42: Missing ini.h file

amigafan, 2018-01-30:
Maybe this is just me being stupid, but where does ini.h come from?
When I try to build I got this error:
ESP32_MP3_Decoder/components/playlist/./playlist.c:19:17: fatal error: ini.h: No such file or directory
And I have tried: "git submodule init && git submodule update", in the esp-idf directory
Any suggestion is much appreciated! Thanks in advance.

MrBuddyCasino, 2018-01-30:
Try executing "git submodule init && git submodule update" in the project directory, not esp-idf.

Von meinem iPhone gesendet

> Am 30.01.2018 um 21:45 schrieb amigafan <notifications@github.com>:
> 
> "git submodule init && git submodule update"

amigafan, 2018-01-31:
When I try I get this error: fatal: Not a git repository (or any of the parent directories): .git

MrBuddyCasino, 2018-01-31:
Well I guess you are in the .git directory? You can find out via `pwd`. If so. go one level higher with `cd ..`.

amigafan, 2018-01-31:
[image]
 Still no luck

MrBuddyCasino, 2018-01-31:
I'm afraid I have no idea. I would try to retry a `git clone` and repeat your steps. This looks like a git issue so I'm closing this for now.

closed 2018-01-31 by MrBuddyCasino