MrBuddyCasino/ESP32_MP3_Decoder #43: Bluetooth Speaker failed to pair

vibnwis, 2018-02-12:
Hi there,
Would very appreciate if you could render some help there. I compiled the code successfully, however, when I tried to connect/pair it. It failed. Any idea why? Here are the message.

I (241) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (521) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0
I (591) renderer: init I2S mode 0, port 0, 16 bit, 44100 Hz
I (591) I2S: queue free spaces: 1
I (601) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=32
I (601) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, divb: 11
I (611) main: RAM left 119532
I (611) bt_speaker: starting "ESP32_BLE_SPEAKER"
E (11661) BT: btm_sec_conn_req

E (11671) BT: btm_sec_connected

MrBuddyCasino, 2018-02-14:
btm_sec_connected implies there was a connection?

KevinCocco, 2018-02-15:
I have the same problem. I paired it successfully but I can't connect to it. The main point on my side is that I could connect to it before but I lost my power connection once and since that I couldn't connect anymore. Any solutions?

MrBuddyCasino, 2018-02-15:
What happens if you unpair (from your laptop or whatever), then pair again?

KevinCocco, 2018-02-15:
I tried the following: Unpair then repairing, Unpluged from Laptop and connected to battery power, Connecting with another device. Nothing worked.

vibnwis, 2018-02-15:
Actually, i managed to get it working by changing another router which has
weaker signals and farther. The one that has issue has strong signal and
nearer is commercial grade AP. This is probably not helpful in many ways.
Cheers!

MrBuddyCasino, 2018-02-16:
What does a router have to do with Bluetooth?

vibnwis, 2018-02-16:
Apologies. My bad!

closed 2018-06-29 by MrBuddyCasino