MrBuddyCasino/ESP32_MP3_Decoder #46: Bluetooth speaker speaker no sound came out

LilyGO, 2018-02-26:
Bluetooth has been successfully connected, and the serial port has been read music information. Pin has also been confirmed, but not the slightest sound.
[image]
[image]
[image]

MrBuddyCasino, 2018-02-28:
I can't make out the pins by the photos. Console output looks fine.

LilyGO, 2018-03-01:
PIN 25 LRCK
PIN 26 BCLK
PIN 22 DATA

LilyGO, 2018-03-01:
[image]

LilyGO, 2018-03-01:
I bought two 98357A last month and one last week. So I think my horn is too small, so I bought a horn. But unfortunately, from beginning to end a trace of Bluetooth device music came out.

closed 2018-06-30 by MrBuddyCasino