MrBuddyCasino/ESP32_MP3_Decoder #47: BLuetooth speaker in crash

donny681, 2018-03-16:
I don't know why the esp32 bluetooth speaker must be in crash when the module get  audio data over 15500 .The ESP32 break down .Can you show me your  menuconfig about  ESP32-specific and Bluetooth ?
[image]

akohlsmith, 2018-04-03:
I see the same problem but always after 700 packets. I'm playing audio from either my iphone 5S or my 11" macbook air

MrBuddyCasino, 2018-04-03:
Got crash logs? The screenshot cut off the interesting parts.

akohlsmith, 2018-04-03:
Yes, I lost them when my computer crashed but will upload them tonight. There doesn't appear to be anything super interesting but I'm still a newb at the ESP32-side of things.

akohlsmith, 2018-04-03:
I wonder if this is related to #14...

akohlsmith, 2018-04-06:
Well it's not exactly a crash log, but it's all that comes out from the serial console:

```
I (554) BTDM_INIT: BT controller compile version [8c18895]

I (554) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (634) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (844) renderer: init I2S mode 3, port 0, 16 bit, 44100 Hz
I (844) I2S: queue free spaces: 1
I (854) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=32
I (854) I2S: PLL_D2: Req RATE: 44100, real rate: 200000.000, BITS: 16, CLKM: 5, BCK: 5, MCLK: 5.669, SCLK: 6400000.000000, diva: 64, divb: 42
I (864) bt_speaker: starting "ESP_SPEAKER"
I (884) main: RAM left 120456
E (158594) BT: bta_av_rc_create ACP handle exist for shdl:0
I (158604) BT_AV: a2dp conn_state_cb, state 1
I (158804) BT_AV: a2dp audio_cfg_cb , codec type 0
I (158804) I2S: PLL_D2: Req RATE: 44100, real rate: 200000.000, BITS: 16, CLKM: 5, BCK: 5, MCLK: 5.669, SCLK: 6400000.000000, diva: 64, divb: 42
I (158814) BT_AV: configure audio player 22-15-2-28

I (158824) BT_AV: audio player configured, samplerate=44100
I (158864) BT_AV: a2dp conn_state_cb, state 2
I (158884) BT_AV: avrc conn_state evt: state 1, [98:fe:94:4d:fb:93]
I (158924) BT_AV: avrc remote features 249
E (180364) BT: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

E (180364) BT: BTA_AV_START_EVT status 0, suspending 0, init 0

I (180374) BT_AV: a2dp audio_state_cb state 2
E (185954) BT: BTA_AV_SUSPEND_EVT status 0, init 0

I (185954) BT_AV: a2dp audio_state_cb state 0
E (189694) BT: bta_av_link_role_ok hndl:x41 role:1 conn_audio:x1 bits:1 features:x824b

E (189694) BT: BTA_AV_START_EVT status 0, suspending 0, init 0

I (189694) BT_AV: a2dp audio_state_cb state 2
E (192874) BT_AV: audio data pkt cnt 100
E (196044) BT_AV: audio data pkt cnt 200
E (199204) BT_AV: audio data pkt cnt 300
E (202374) BT_AV: audio data pkt cnt 400
E (205534) BT_AV: audio data pkt cnt 500
E (208704) BT_AV: audio data pkt cnt 600
E (211864) BT_AV: audio data pkt cnt 700
```

MrBuddyCasino, 2018-04-06:
Hmm. That smells like an issue in the underlying BT software stack, and I don't have the expertise to interpret those messages.

hemanth7787, 2018-08-20:
### Problem:
Crash and plays looping glitch sound.

### Build config
[sdkconfig.pdf](https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/files/2302864/sdkconfig.pdf)

### Log
[image]

### full log
[log.txt](https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/files/2302867/log.txt)