MrBuddyCasino/ESP32_MP3_Decoder #48: return MAD_ERROR_LOSTSYNC when decode mp3 file, lost synchronization

NOA-Vernon, 2018-03-30:
Hi,
I want to read the mp3 file from sd card, decode it and output throuhg built-in DAC, but it occured an error(MAD_ERROR_LOSTSYNC ).
following is my code that calling funcion: 
static void play_mp3_file(void *vparameter)
{
	 web_radio_t *radio_conf = vparameter;
	 
	char mp3_data[1024];//[4096];

	
	player_t *player_config = radio_conf->player_config;
	
 	int ret;
 	int i;

ESP_LOGI(TAG,"play_mp3_file task\r\n");
	FILE* f = fopen("/sdcard/ghsy.mp3","r");
	if (f == NULL) {
		ESP_LOGI(TAG,"Failed to open file for reading.\r\n");
		vTaskDelete(NULL);
	}

    player_config->media_stream->content_type = AUDIO_MPEG;
    player_config->media_stream->eof = false;
	
player_config->command = CMD_START;
player_config->buffer_pref = BUF_PREF_FAST;//BUF_PREF_SAFE;//

    audio_player_start(player_config); 
	while(1)
	{
		ret=fread(mp3_data,1,sizeof(mp3_data),f);
		
		if(ret <= 0)
		{
			ESP_LOGI(TAG,"play mp3 read file error.\r\n");
			break;
		}
ESP_LOGI(TAG,"play_mp3_file read data succ\r\n");
		while( audio_stream_consumer(mp3_data, sizeof(mp3_data), player_config) != 0)
		{
			vTaskDelay(500/ portTICK_PERIOD_MS);
			ESP_LOGI(TAG,"play_mp3_file vTaskDelay ing\r\n");
		}
		vTaskDelay(3000/ portTICK_PERIOD_MS);
	}
	fclose(f);
	printf("delete play_mp3_file_task.\r\n");
	vTaskDelete(NULL);
}

void web_radio_start(web_radio_t *config)
{
    // start reader task
    //xTaskCreatePinnedToCore(&http_get_task, "http_get_task", 2560, config, 20,
    //NULL, 0);
ESP_LOGI(TAG,"web_radio_start\r\n");	
	xTaskCreate(&play_mp3_file, "play_mp3_file", 4096+4096, config, 5, NULL);

}

and the debug log is:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
[... 217 lines cut ...]
[0;31mE (84739) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84749) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84749) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84759) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84769) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84769) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84779) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84779) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84789) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84799) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84799) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84809) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84809) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84819) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84829) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84829) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84839) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84849) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84849) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84859) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84859) mad_decoder: Buffer underflow, need 2881 bytes.[0m
[0;31mE (84869) mad_decoder: Buffer underflow, need 2881 bytes.[0m

Is there any hint or solution for solving this problem?
thank you in advance.

SinanAkkoyun, 2020-01-27:
Did you figure it out? I am also working on it...

NOA-Vernon, 2020-02-10:
I just increased the mp3 decode buffer for getting more data from remote server before playing, but  this is a workaround only.