MrBuddyCasino/ESP32_MP3_Decoder #51: Start a wiki with compatible hardware?

tonyp7, 2018-05-04:
**Note: not an actual issue.**

It would be nice to have a wiki showing up various compatible hardwares and project derived from your code. 

Right now I am building a breakout for the [TAS5733L](http://www.ti.com/lit/ds/symlink/tas5733l.pdf) because the MAX chip is very limited. Will it work with this code? I have no idea. Theoretically it should, but having a clear wiki that says "YES" would be a great resource. 

Cheers,
Tony

MrBuddyCasino, 2018-05-04:
I2S chip compatibility is more of a hardware issue, unless some I2C configuration stuff is required. If you want to make a list of devices, you're welcome to do so - just make a pull request.

tonyp7, 2018-05-10:
Cool -- I'll make sure to do that. My PCB is in transit somewhere and I'll be off two weeks due to work constraints, but I'll make sure to post results here!
See: https://idyl.io/2018/05/07/tas5733l-audio-amplifier/

tonyp7, 2018-06-02:
https://imgur.com/zNhvjp3

Well I am reporting in to say that so far I have a pretty board but I can't make this sucker work. I am still trying to figure out what I am doing wrong, but in the meantime TI released a new chip (http://www.ti.com/product/tas5805m) that is really exciting. It's basically a simplified version of the TAS5733L. They removed MCLK completely and managed to make it inductor less. 

I'll keep digging on the TAS5733L though... Damn it's frustrating when you have something that straight doesn't work like that!

MrBuddyCasino, 2018-06-02:
I know that feel :-/
Good luck, all it takes is persistence and an oscilloscope.

tonyp7, 2018-08-04:
Happy to report the TAS5805M is working! I'm adjusting a few things but I'll make sure to post results!

tonyp7, 2018-08-09:
There it is!
https://idyl.io/tas5805m-audio-amp-i2s.webm