MrBuddyCasino/ESP32_MP3_Decoder #53: Many warnings at compile time (esp-idf 3.0)

tonyp7, 2018-05-29:
Most notably deprecated functions:

**Deprecated functions:**
C:/msys32/home/tony/esp/ESP32_MP3_Decoder/components/audio_renderer/audio_renderer.c:133:13: warning: 'i2s_write_bytes' is deprecated [-Wdeprecated-declarations]
             bytes_written = i2s_write_bytes(renderer_instance->i2s_num, buf, bytes_left, 0);

C:/msys32/home/tony/esp/ESP32_MP3_Decoder/components/audio_renderer/audio_renderer.c:182:13: warning: 'i2s_push_sample' is deprecated [-Wdeprecated-declarations]
             bytes_pushed = i2s_push_sample(renderer_instance->i2s_num, (const char*) &sample, max_wait);

But also a tons of other things: https://pastebin.com/HJuW5efB

MrBuddyCasino, 2018-05-29:
Wanna make a pull request?

tonyp7, 2018-05-29:
Sure, let me try my audio amp first (finally soldered it...) and I'll try cleaning up some code and see if we can reduce this warning count!