MrBuddyCasino/ESP32_MP3_Decoder #54: I2S seems to work, But no sound

jodel477, 2018-06-12:
Hello guys,
i just flashed my esp32 with this great Software, but I want to use it with Hifiberry Digi (WM 8804 SPDIF IC). The I2S transmission seems to work as far as I can See with my oscilloscope, but I don't get SPDIF Signals. The WM8804 board work quite good with my Raspberry Pi, but not with the ESP.  Do you know if i need to configure something else? Please help me...

jodel477, 2018-06-12:
Btw: Bitclock, Lrclk and Data are connected correctely.  I'm an I2S newbie... Thanks in advance!

MrBuddyCasino, 2018-06-12:
Its possible the WM 8804 needs MCLK. ESP32 does not natively generate MCLK, best to search the esp32.com forum for solutions. Please report back when you found one.

jodel477, 2018-06-12:
Thanks for your fast answer, but the RPi doesn't generate the MCLK, too. So I think it should work without MCLK...

MrBuddyCasino, 2018-06-12:
I see, in that case I have no idea.

jodel477, 2018-06-13:
Hi guys,
i found out, that the Digi+ is the Master. So the esp32 needs to be the slave, but when configuring "I2S_MODE_SLAVE | I2S_MODE_TX", i get an " Bluetooth Packet Drop" Error (thousands per second) via UART. Maybe you can help me with this information?

MrBuddyCasino, 2018-06-13:
Never used I2S slave mode, sorry.

MrBuddyCasino, 2018-06-29:
Closing due to inactivity.

closed 2018-06-29 by MrBuddyCasino