MrBuddyCasino/ESP32_MP3_Decoder #55: Buffer underflow issue

medamarko, 2018-06-26:
Constantly getting buffer underflow.... Im using ESP Wroom 32 module. Do i need Wrover with PSram to run this app?

` (164423) mad_decoder: Buffer underflow, need 2471 bytes`

khabte, 2018-08-10:
I ran into the exact same issue.

Calling [i2s_write](https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/i2s.html#_CPPv29i2s_write10i2s_port_tPKv6size_tP6size_t10TickType_t) instead of the deprecated function [i2s_push_sample](https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/i2s.html#_CPPv215i2s_push_sample10i2s_port_tPKv10TickType_t) seems to fix it for me. 

https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/blob/580bbb94a4d0c0729af50d942941a616f925f93a/components/audio_renderer/audio_renderer.c#L194

I'll create a PR with the changes.

myzhan, 2018-08-18:
I met the same issue, and this patch solved it.

PR #59 (2018-08-26): Changed 'i2s_push_sample' and 'i2s_write_bytes' to 'i2s_write'.
This PR should fix #55.

I'm fairly new to the ESP32 and contributing on Github, so please let me know if I should change or improve something.

Did you set the 'ticks_to_wait' variable to 0 on purpose, or is it okay to change it to 'max_wait' like the other i2s_write calls?

https://github.com/MrBuddyCasino/ESP32_MP3_Decoder/blob/580bbb94a4d0c0729af50d942941a616f925f93a/components/audio_renderer/audio_renderer.c#L133
closed 2018-08-26 by MrBuddyCasino