MrBuddyCasino/ESP32_MP3_Decoder #57: bt_speaker mode doesn't compile

tonyp7, 2018-08-06:
So I've been playing with the code mostly using web radio. Today I wanted to try the bluetooth speaker. I downloaded a fresh copy of the code and got this message:

```
ESP32_MP3_Decoder_clean/components/bt_speaker/bt_speaker.c:93:9: error: implicit declaration of function 'esp_a2d_sink_register_data_callback' [-Werror=implicit-function-declaration]
         esp_a2d_sink_register_data_callback(bt_app_a2d_data_cb);
         ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/tony/esp/esp-idf-v3.0/make/component_wrapper.mk:274: bt_speaker.o] Error 1
make: *** [C:/msys32/home/tony/esp/esp-idf-v3.0/make/project.mk:450: component-bt_speaker-build] Error 2
```

Seems like you can't compile the code anymore with just "make" you have to change the default esp-idf config (under make/project.mk) and set the following gcc flag:

-Wno-error=implicit-function-declaration

And this is a pretty scary flag to include!

MrBuddyCasino, 2018-08-06:
Thanks for the report, looks like that needs to be fixed. PR?

tonyp7, 2018-08-06:
Unfortunately no, I'm not quite sure what is going wrong. I compiled esp-idf bluetooth speaker example (https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_sink) to be sure I don't have an issue specific to my setup but this one compiles with no warnings at all.

aliafshar, 2018-08-06:
I don't know if what I did was right, but that method doesn't exist in the sdk version I have here. The PR I made works for me, anyway.

mixographer, 2018-08-08:
Now I get this one. 

```
/Users/jimmy/git/ESP32_MP3_Decoder/components/bt_speaker/bt_speaker.c: In function 'bt_av_hdl_stack_evt':
/Users/jimmy/git/ESP32_MP3_Decoder/components/bt_speaker/bt_speaker.c:93:9: error: implicit declaration of function 'esp_a2d_register_data_callback' [-Werror=implicit-function-declaration]
         esp_a2d_register_data_callback(bt_app_a2d_data_cb);
         ^
cc1: some warnings being treated as errors
make[1]: *** [bt_speaker.o] Error 1
```
Is there a mismatch between the SDK and the latest version?

tonyp7, 2018-08-09:
The latest a2dp sink example from esp-idf 3.0.2 (latest stable release) has both:

MrBuddyCasino, 2018-08-09:
Yep, I reverted the change because thats not the issue. It seems the build can't see the a2dp headers, I still don't quite understand why.

MrBuddyCasino, 2018-08-09:
added failsafe to kconfig

closed 2018-08-09 by MrBuddyCasino