MrBuddyCasino/ESP32_MP3_Decoder #63: Bluetooth as a audio output

jgrulich, 2018-09-22:
How to make the Bluetooth as a audio output? I was searching in the esp32 wiki and found that the Bluetooth may be used as master to source the audio to the external Bluetooth speaker, but I've not found this feature in the menuconfig.

zorglups, 2019-01-02:
I'm also looking for a way to play the audio over bluetooth to listen using my bluetooth headset.
Is that possible ?

MrBuddyCasino, 2019-01-02:
Its possible but not implemented. You'd just have to re-reroute the output in the AudioRenderer.

aadnesd, 2019-06-28:
Hi, sorry to hijack, has this been implemented now?

MrBuddyCasino, 2019-06-28:
No PRs as of yet. It shouldn't be hard, but I don't currently have the time.

aadnesd, 2019-07-30:
Could you give me some pointers, and I'll try to get something to work?

dgaspar, 2019-11-27:
This example might shed some light on how to render the audio to an external bluetooth speaker: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/a2dp_source

nathanRamaNoodles, 2019-12-09:
Any progress?

Beefy-Swain, 2020-07-14:
I am very interested in this as well

PapeCoding, 2021-08-31:
I am also interested in this!