MrBuddyCasino/ESP32_MP3_Decoder #65: disconnected for a while

NimaaZx, 2018-12-20:
Hello Buddy. I'm using your clone code ( customized buffer size- this version: https://github.com/esa-n/ESP32_MP3_Decoder)
maybe there is a similar problem in your code. 
I need your help to understand what is that.

I use esp32 in local network for stream audio from Icecast2 and mixxx player to play mp3.
sometime i get this error and it seems esp32 disconnected for a few seconds.

Error:
"I (754031) http_client: ... done reading from socket. Last read return=-1 errno=
11"

it is necessary for me to receive every single byte and never disconnected.
I can donate for you.
tnx a lot.

MrBuddyCasino, 2018-12-20:
Looks like a read timeout. Does it restart streaming automatically or stop there?

NimaaZx, 2018-12-20:
thanks a lot .No it is restarted. after a few seconds streaming is going to be ok. i use built in dac.
recently i have bought a PCM5102A for external I2S .maybe it is for that. i have not tried it yet.
from internet radio server I received that error. in webradio using it is not very important. but in my diy project is important.
```
...
I (742993) audio_player: Buffer fill 81%, 65521 bytes RAM left 85200
I (746566) audio_player: Buffer fill 87%, 69660 bytes RAM left 87972
I (750345) audio_player: Buffer fill 83%, 66401 bytes RAM left 90072
I (754031) http_client: ... done reading from socket. Last read return=-1 errno=
11
I (754032) http_client: socket closed
I (754032) web_radio: http_client_get completed
I (754038) http_client: DNS lookup succeeded. IP=192.168.1.100
I (754045) http_client: ... allocated socket
I (754656) http_client: ... connected
I (754656) http_client: requesting GET /stream HTTP/1.0
Host: 192.168.1.100:8000
User-Agent: ESP32
Accept: */*

I (754662) http_client: ... socket send success
E (755463) mad_decoder: dec err 0x0101 (lost synchronization)
E (755464) mad_decoder: dec err 0x0235 (bad main_data_begin pointer)
I (755499) audio_player: Buffer fill 53%, 42904 bytes RAM left 81964

```
I use this mount and url for broadcasting.
192.168.1.100:8000/stream

do you work with another module or something with higher level programming language that is similar your project without any disconnected or interrupted?
I thought there is a Esp32 watchdog or some hardware limitation.
What is your suggest for broadcasting digital audio? 
one sender and many receivers. I see raspberrypi diy but it is a little expensive for me.

MrBuddyCasino, 2018-12-20:
> What is your suggest for broadcasting digital audio? one sender and many receivers.

Thats hard to say without additional context. What is it that you want to do? If you want a dependable high-level solution, the Raspberry Pie Zero is actually quite nice.

NimaaZx, 2018-12-20:
OK.There is a Educational English Class. Right Now It is Simulating IELTS Listening with wired headphone.
I want to use a wireless system with stereo output. I have seen other RF modules that use radio frequency.quality is not as good as stereo or 128kbps MP3.
As I Searched before Raspberry Pie Zero does not have wifi or dac and aux And I have a  Budget constraint. 
do you think what component I should use with Raspberry Pie Zero?

MrBuddyCasino, 2018-12-20:
The Pie Zero W does have wifi.
But if you're doing broadcast, there is a limit to how many clients you can support. I'd use wires or real broadcast, or Wifi multicast if all esle fails. You might want to check out the ESP-NOW protocol for that.

NimaaZx, 2018-12-20:
about client limitation, i think it is a limitation in network bandwidth topic.
for example using a server for each group. (20 recievers)
all in all do you think it is possible to use esp32 for this situation without any packet drops?
do you suggest using another hardware good for this project?

as i searched esp now is good for peer to peer connection.

NimaaZx, 2018-12-24:
I connected I2S module and disconnecting is still there. I forgot to ask you, maybe there is a limitation memory buffer or heap size. most of the time I starting streaming Fill buffer about 85 and 90 % and going to 99% . it is not started from 10 or 20%. 
I changed many things. such as : SPIRAMSIZE , SPIREADSIZE , char recv_buf[].
Not worked for my problem.

MrBuddyCasino, 2018-12-24:
SPIRAMSIZE and wifi conditions will be the biggest factors. You'll want a module with a decent antenna, not all of them are well designed. Which one you've got?

NimaaZx, 2018-12-24:
I'm Sorry but i can't what you said. I use just two modules.
ESP32
I2S PCM5102 I2S module. 
Esp32 does not have wifi antenna and i think it is not related to my problem. it is near to wifi modem.
a few days ago i wrote error number message , you said it seems time out.
at this moment i get a lot of  this error in Make monitor and sound is gone for a few seconds like previous error i wrote a few days a ago and i think it is more important.
```
I (47643) audio_player: Buffer fill 98%, 78915 bytes RAM left 72844
I (55665) audio_player: Buffer fill 97%, 78250 bytes RAM left 72844
I (63529) audio_player: Buffer fill 99%, 79498 bytes RAM left 76280
E (76268) mad_decoder: Buffer underflow, need 2532 bytes.
I (76269) mad_decoder: decoder stopped
I (76269) mad_decoder: MAD decoder stack: 2020

I (76293) mad_decoder: decoder start
E (76293) mad_decoder: Buffer underflow, need 2889 bytes.
I (76293) mad_decoder: decoder stopped
I (76296) mad_decoder: MAD decoder stack: 2020

I (76702) http_client: ... done reading from socket. Last read return=-1 errno=1
1
I (76705) http_client: socket closed
I (76706) web_radio: http_client_get completed
I (76712) http_client: DNS lookup succeeded. IP=96.44.145.66
I (76717) http_client: ... allocated socket
I (77128) http_client: ... connected
I (77130) http_client: requesting GET /977_HITS.mp3 HTTP/1.0
Host: 19763.live.streamtheworld.com:80
User-Agent: ESP32
Accept: */*

I (77137) http_client: ... socket send success
I (78162) audio_player: Buffer fill 5%, 4736 bytes RAM left 97700
I (80622) mad_decoder: decoder start
E (80623) mad_decoder: dec err 0x0235 (bad main_data_begin pointer)
E (80624) mad_decoder: dec err 0x0235 (bad main_data_begin pointer)
I (82897) audio_player: Buffer fill 97%, 78262 bytes RAM left 76152

```

I think we should freed buffer, I read all your issues. it is underflow error and i think it is for lake of buffer. buffer fill is about 98% most of the time.
in another issue you suggested  changing  SPIRAMSIZE  and i changed it to 80000 and changed char recv_buf[ ] to char recv_buf[512] or [256].
Totally all of them not work and underflow error appears.

NimaaZx, 2018-12-24:
I think esp32 is just suitable for school or university projects. not for building a product. many issues and different behaviour are appeared everytime I used.