MrBuddyCasino/ESP32_MP3_Decoder #69: Why not use esp-adf?

Motherboard, 2019-06-12:
Hi, I'm starting an mp3 player project on esp32 (not webradio, a portable mp3 player). 

Your project seems very relevant as a starting point, but I've also seen esp-adf, which seems also very promising as a starting point, and provides aac in addition to mp3. 

Is there a good reason not to use esp-adf?

MrBuddyCasino, 2019-06-12:
This project precedes ADF. AAC is supported, too btw.

Motherboard, 2019-06-12:
Thanks. 

Would you recommend forking your project and adding the OLED and microSD card support, or going with ADF? I also need to add fast forward (faster playback) and seek functionality (for rewind and when the fast forward needs to be faster than what it'd make sense to playback).

If you could give me some pointers, that'd be great :)

MrBuddyCasino, 2019-06-12:
There might be forks that already added OLED, and maybe microSD. I didn't look at ADF yet, so I can't honestly tell.

closed 2019-06-12 by Motherboard