MrBuddyCasino/ESP32_MP3_Decoder #71: No bluetooth speaker mode in menuconfig

aadnesd, 2019-06-28:
Hi, there is no bluetooth speaker mode in menuconfig. I got to enable classic BT and A2DP in component config, but there is no BT speaker mode anywhere. I looked in the Kconfig file and it looks like it should definitely be under "Web Radio/BT Speaker", and it also looks like the BT_SPEAKER MODE part is wrong when comparing to WIFI Config and Audio Output which I'm able to see in menuconfig. I've tried to change the Kconfig file, but I'm not able to do so and still run make menuconfig.

aadnesd, 2019-06-28:
I eventually got the option of selecting BT_SPEAKER_MODE after just trying different things in the Kconfig. However, now I get a whole lot of errors when I try to flash. I see that BT_SPEAKER_MODE has been activated as the first error is in bt_speaker.h.

bastiffi, 2019-07-06:
I have the same problem. On Mac OS 10.13.4.

vedranMv, 2019-07-08:
Same problem here. Menu is not available. Any idea what you made to get it?

vedranMv, 2019-07-09:
Got it working. Here are the steps:

- Replace esp-idf with v3.2.2 (as per #67 ): https://github.com/espressif/esp-idf/releases (just download .zip and unpack it instead of your current esp-idf)

- Run `make menuconfig`

- Component config->Bluetooth->Bluetooth controller->Bluetooth controller mode (enable dual mode)
(Otherwise you get error message `BT_AV: bt_speaker_start enable controller failed` on the serial port during startup)
[image]

- Component config->Bluetooth->Bluedroid enable->Classic Bluetooth & A2DP

- Web radio / Bluetooth speaker -> Bluetooth speaker mode (now visible)

pallsopp, 2020-02-10:
As far as I noticed there was no BT speaker mode until after I selected Classic Bluetooth, and enabled A2DP. Then go back to the Web Radio section and it appears.

danvonk, 2020-04-04:
I had to downgrade to v3.3.1 too as it didn't work on the latest git version of ESP-INF. Not ideal as it's a bit tricky to get working with my distro's python version.