MrBuddyCasino/ESP32_MP3_Decoder #78: Buffer Underflow ESP32 keeps crashing

ShivamJoker, 2020-04-30:
I have esp32S or wroom 32 and it keeps crashing :(
```
E (34712) mad_decoder: Buffer underflow, need 2604 bytes.
E (34722) mad_decoder: Buffer underflow, need 2604 bytes.
E (34722) mad_decoder: Buffer underflow, need 2604 bytes.
E (34732) mad_decoder: Buffer underflow, need 2604 bytes.
E (34742) mad_decoder: Buffer underflow, need 2604 bytes.
E (34742) mad_decoder: Buffer underflow, need 2604 bytes.
E (34752) mad_decoder: Buffer underflow, need 2604 bytes.
I (34752) http_client: ... done reading from socket. Last read return=0 errno=128
I (34772) http_client: socket closed
I (34772) web_radio: http_client_get completed
I (34822) mad_decoder: decoder stopped
I (34822) mad_decoder: MAD decoder stack: 2484
```

MrBuddyCasino, 2020-04-30:
Your connection is probably bad / too slow. Might need to fix our ESP32 antenna or move it closer to the AP.

ShivamJoker, 2020-04-30:
oh okay got it I connected to my mobile hotspot and now it works fine but PDM seems to generate noise and dac is kinda okay

MrBuddyCasino, 2020-04-30:
I2S should work, nobody figured out PDM yet afaik

ShivamJoker, 2020-04-30:
okay thanks a lot for this project

closed 2020-04-30 by ShivamJoker

dmitrij9992905, 2020-06-19:
I have the same issue, but I use external DAC and good internet connection. I assembled my project on esp32-wroom.

here are logs:

The first time after booting player plays smoothly
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6816
ho 0 tail 12 room 4
load:0x40078000,len:12072
ho 0 tail 12 room 4
load:0x40080400,len:6708
entry 0x40080778
I (74) boot: Chip Revision: 1
I (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v3.3.2-107-g722043f73 2nd stage bootloader
I (42) boot: compile time 09:51:16
I (42) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00200000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x583a4 (361380) map
I (240) esp_image: segment 1: paddr=0x000683cc vaddr=0x3ffbdb60 size=0x05900 ( 22784) load
I (250) esp_image: segment 2: paddr=0x0006dcd4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/dmitrij999/esp/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1779

I (250) esp_image: segment 3: paddr=0x0006e0dc vaddr=0x40080400 size=0x01f34 (  7988) load
I (262) esp_image: segment 4: paddr=0x00070018 vaddr=0x400d0018 size=0xb2db0 (732592) map
0x400d0018: _stext at ??:?

I (524) esp_image: segment 5: paddr=0x00122dd0 vaddr=0x40082334 size=0x14a90 ( 84624) load
0x40082334: shared_intr_isr at /home/dmitrij999/esp/esp-adf/esp-idf/components/esp32/intr_alloc.c:784

I (574) boot: Loaded app from partition at offset 0x10000
I (574) boot: Disabling RNG early entropy source...
I (574) cpu_start: Pro cpu up.
I (578) cpu_start: Application information:
I (583) cpu_start: Project name:     mp3_decoder
I (588) cpu_start: App version:      fe57c88-dirty
I (593) cpu_start: Compile time:     Jun 19 2020 09:51:18
I (599) cpu_start: ELF file SHA256:  27d79fb5191c9f44...
I (605) cpu_start: ESP-IDF:          v3.3.2-107-g722043f73
I (612) cpu_start: Starting app cpu, entry point is 0x40081258
0x40081258: call_start_cpu1 at /home/dmitrij999/esp/esp-adf/esp-idf/components/esp32/cpu_start.c:268

I (0) cpu_start: App cpu up.
I (622) heap_init: Initializing. RAM available for dynamic allocation:
I (629) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (635) heap_init: At 3FFD2FC0 len 0000D040 (52 KiB): DRAM
I (641) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (648) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (654) heap_init: At 40096DC4 len 0000923C (36 KiB): IRAM
I (660) cpu_start: Pro cpu start user code
I (8) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (9) main: starting app_main()
I (9) main: RAM left: 223172
[... 167 lines cut ...]
E (18949) mad_decoder: Buffer underflow, need 2732 bytes.
E (18949) mad_decoder: Buffer underflow, need 2732 bytes.
E (18949) mad_decoder: Buffer underflow, need 2732 bytes.
E (18959) mad_decoder: Buffer underflow, need 2732 bytes.
E (18969) mad_decoder: Buffer underflow, need 2732 bytes.
E (18969) mad_decoder: Buffer underflow, need 2732 bytes.
E (18979) mad_decoder: Buffer underflow, need 2732 bytes.
E (18979) mad_decoder: Buffer underflow, need 2732 bytes.
E (18989) mad_decoder: Buffer underflow, need 2732 bytes.
E (18999) mad_decoder: Buffer underflow, need 2732 bytes.
E (18999) mad_decoder: Buffer underflow, need 2732 bytes.
E (19009) mad_decoder: Buffer underflow, need 2732 bytes.
E (19009) mad_decoder: Buffer underflow, need 2732 bytes.
E (19019) mad_decoder: Buffer underflow, need 2732 bytes.
E (19029) mad_decoder: Buffer underflow, need 2732 bytes.
E (19029) mad_decoder: Buffer underflow, need 2732 bytes.
E (19039) mad_decoder: Buffer underflow, need 2732 bytes.
...
E (19309) mad_decoder: Buffer underflow, need 2604 bytes.
E (19319) mad_decoder: Buffer underflow, need 2604 bytes.
E (19319) mad_decoder: Buffer underflow, need 2604 bytes.
E (19329) mad_decoder: Buffer underflow, need 2604 bytes.
```

Is it memory leak?

dmitrij9992905, 2020-06-19:
I added 256kbps station to playlist. It's being played smoothly. Intermittance is observed on bitrates 128 and lower