MrBuddyCasino/ESP32_MP3_Decoder #79: AAC in bluetooth mode?

HITMAnsOFT, 2020-05-01:
Hi, is there support for AAC codec in A2DP mode? It would be very interesting for making DIY high quality Bluetooth DACs with AAC support, given that AptX is not open and AAC decoder is already included.

MrBuddyCasino, 2020-05-01:
I had a look at this, unfortunately no, its not extensible. Can be hacked maybe by patching the bt stack, but I didn’t try.

MrBuddyCasino, 2020-05-01:
A high bitrate SBC is already very good, the bad reputation is unwarranted. Some early devices (eg Android) capped the max bitrate for some dumb reason, which made it sound bad, but I don’t think current devices fo that anymore.

HITMAnsOFT, 2020-05-02:
So if I understand correctly, SBC advertising and decoding is fully handled by BT stack in esp-idf. Thanks for the answer!

closed 2020-05-02 by HITMAnsOFT